ros-controls/gz_ros2_control #362: Cannot locate rosdep definition for [gz_plugin_vendor]

MnM03, 2024-07-05:
Hello I am trying to install ros_ign from source, and I receive the following error:

root@mohamed-IdeaPad-Pro-5-14APH8:~/ros_ign/src# rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
gz_ros2_control: Cannot locate rosdep definition for [gz_plugin_vendor]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

How can I solve this?

christophfroehlich, 2024-07-05:
On which operating system are you working, and which ROS distro?

august779188, 2024-07-09:
> On which operating system are you working, and which ROS distro?您使用哪个操作系统以及哪个 ROS 发行版？

I use humble version and i meet the same question. I also find that gz_plugin_vendor seems only support rolling and jazzy version

christophfroehlich, 2024-07-12:
@august779188 Are you using the correct branch, as listed [here](https://github.com/ros-controls/gz_ros2_control?tab=readme-ov-file#gz_ros2_control)?
For humble, you need the `humble` branch which has no dependency on `gz_*_vendor` packages. If you are instead trying to compile the rolling version of ros2_control stack, there is no way to do so on ubuntu jammy.

MnM03, 2024-07-12:
On which operating system are you working, and which ROS distro?

Dear Christoph,

I apologize for the late reply. I managed to install the plugin by using apt-install command. I first used apt-get, which did not work. So I tried to buy locally. I am using a docker container with humble: osrf/ros:humble-desktop-full.

MnM03, 2024-07-12:
> On which operating system are you working, and which ROS distro?

Dear Christoph,

I aplogize for the late reply. I managed to install the plugin by using apt install command. I first used apt-get, which did not work. So I tried to buid locally. I am using a docker container with humble: osrf/ros:humble-desktop-full.

christophfroehlich, 2024-07-12:
As I said before, if you are using humble and want to build ign_ros2_control from source, you have to use the `humble` branch here.

closed 2024-07-12 by christophfroehlich

MnM03, 2024-07-12:
> As I said before, if you are using humble and want to build ign_ros2_control from source, you have to use the `humble` branch here.

And I did.. I selected the 'humble' branch.

christophfroehlich, 2024-07-12:
I don't think so.

- In humble there was no gz_ros2_control, it is called ign_ros2_control.
- There is no dependency on gz_plugin_vendor on humble: https://github.com/ros-controls/gz_ros2_control/blob/humble/ign_ros2_control/package.xml