ros-controls/gz_ros2_control #372: Add ros2_control tag and ign_ros2_control plugin to a URDF

august779188, 2024-07-09:
When I run this command:**ros2 launch ign_ros2_control_demos cart_example_effort.launch.py**, 
the whole program crashes and the log output is  ign gazebo server:

-  symbol lookup error: /opt/ros/humble/lib/libign_ros2_control-system.so: undefined symbol: _ZN18hardware_interface15ResourceManager9load_urdfERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEbb

- [ERROR] [ros2-4]: process has died [pid 26055, exit code 2, cmd 'ros2 control load_controller --set-state active joint_state_broadcaster'].

Is there a solution?

christophfroehlich, 2024-07-12:
Which operating system are you using, and how did you install ign_ros2_control?

andreacelani, 2024-11-10:
I fixed with:
`sudo apt-get update`
`sudo apt-get upgrade`

and then compiling from source.

I also added:
export IGN_GAZEBO_SYSTEM_PLUGIN_PATH=/opt/ros/humble/lib/
export GZ_SIM_SYSTEM_PLUGIN_PATH=/opt/ros/humble/lib/

to .bashrc

hope this help

Sathiyaseelan10, 2024-12-17:
the solution not workout for me.is there another solution for this

andresgaluitcl, 2025-06-26:
I have the same issue.
I'm migrating from Gazebo Classic to Fortress, on Ubuntu 22.04.5, with ROS2 Humble installed. ROS2 control installed using `sudo apt install ros-{DISTRO}-gz-ros2-control`

When modifying the Gazebo ros2 plugin, I landed on this version:
```
<gazebo>
        <plugin filename="libign_ros2_control-system.so" name="ign_ros2_control">
            <robot_param>robot_description</robot_param>
            <robot_param_node>robot_state_publisher</robot_param_node>
            <parameters>$(find pixkit_control)/config/pixkit_controller_sim.yaml</parameters>
        </plugin>
    </gazebo>
```

But the following error appears when launching the simulation:
```
[ign gazebo-11] Error while loading the library [/opt/ros/humble/lib/libign_ros2_control-system.so]: /opt/ros/humble/lib/libign_ros2_control-system.so: undefined symbol: _ZTIN18controller_manager17ControllerManagerE
[ign gazebo-11] [Err] [SystemLoader.cc:103] Failed to load system plugin [libign_ros2_control-system.so] : couldn't load library on path [/opt/ros/humble/lib/libign_ros2_control-system.so].
```

I started searching and found [this](https://github.com/ros-controls/gz_ros2_control/blob/humble/doc/index.rst), so I tried the demos (the cart_example_velocity in particular), and I get a similar error:
```
ros2 launch gz_ros2_control_demos cart_example_velocity.launch.py

[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [113100]
[INFO] [parameter_bridge-2]: process started with pid [113103]
[INFO] [robot_state_publisher-3]: process started with pid [113105]
[INFO] [create-4]: process started with pid [113107]
[ign gazebo-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[parameter_bridge-2] [INFO] [1750928294.234434417] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (gz.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[robot_state_publisher-3] [WARN] [1750928294.248564198] [kdl_parser]: The root link slideBar has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [INFO] [1750928294.248610775] [robot_state_publisher]: got segment cart
[robot_state_publisher-3] [INFO] [1750928294.248651839] [robot_state_publisher]: got segment imu
[robot_state_publisher-3] [INFO] [1750928294.248655348] [robot_state_publisher]: got segment leg1
[robot_state_publisher-3] [INFO] [1750928294.248658071] [robot_state_publisher]: got segment leg2
[robot_state_publisher-3] [INFO] [1750928294.248660604] [robot_state_publisher]: got segment slideBar
[create-4] [INFO] [1750928294.567368367] [ros_gz_sim]: Requesting list of world names.
[create-4] [INFO] [1750928294.769536075] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-4] [INFO] [1750928294.781957852] [ros_gz_sim]: Requested creation of entity.
[create-4] [INFO] [1750928294.781989611] [ros_gz_sim]: OK creation of entity.
[INFO] [create-4]: process has finished cleanly [pid 113107]
[INFO] [spawner-5]: process started with pid [113290]
[spawner-5] [INFO] [1750928295.402613211] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ign gazebo-1] Error while loading the library [/opt/ros/humble/lib/libgz_ros2_control-system.so]: /opt/ros/humble/lib/libgz_ros2_control-system.so: undefined symbol: _ZTIN18controller_manager17ControllerManagerE
[ign gazebo-1] [Err] [SystemLoader.cc:103] Failed to load system plugin [gz_ros2_control-system] : couldn't load library on path [/opt/ros/humble/lib/libgz_ros2_control-system.so].
[ign gazebo-1] libEGL warning: egl: failed to create dri2 screen
[ign gazebo-1] libEGL warning: egl: failed to create dri2 screen
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[parameter_bridge-2] [INFO] [1750928299.659343296] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-3] [INFO] [1750928299.659343789] [rclcpp]: signal_handler(signum=2)
[INFO] [parameter_bridge-2]: process has finished cleanly [pid 113103]
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 113105]
[INFO] [spawner-5]: process has finished cleanly [pid 113290]
[ERROR] [ign gazebo-1]: process has died [pid 113100, exit code -2, cmd 'ruby /usr/bin/ign gazebo  -r -v 1 empty.sdf --force-version 6'].
[INFO] [launch]: process[ign gazebo-1] was required: shutting down launched system

```

What is the correct plugin to use in my robot description? 
Why does the demo fail too?

christophfroehlich, 2025-06-26:
> ```
> [ign gazebo-11] Error while loading the library [/opt/ros/humble/lib/libign_ros2_control-system.so]: /opt/ros/humble/lib/libign_ros2_control-system.so: undefined symbol: [...]

This looks like an ABI incompatibility. Can you please update all ROS packages (apt-get upgrade) and see if this helps?

andresgaluitcl, 2025-06-26:
> This looks like an ABI incompatibility. Can you please update all ROS packages (apt-get upgrade) and see if this helps?

Yes, that solved the issue with the demos.

As for my code, I had to modify the urdf to this:
```
<gazebo>
        <plugin filename="gz_ros2_control-system" name="gz_ros2_control::GazeboSimROS2ControlPlugin">
            <robot_param>robot_description</robot_param>
            <robot_param_node>robot_state_publisher</robot_param_node>
            <parameters>$(find pixkit_control)/config/pixkit_controller_sim.yaml</parameters>
        </plugin>
    </gazebo>
```
The same as in the demos.

Thanks @christophfroehlich, I appreciate the quick and accurate answer.

closed 2025-09-26 by christophfroehlich