ros-controls/gz_ros2_control #376: ackerman drive controller odometry rotation changes continuously

luxapana, 2024-07-27:
I have setup an ackerman controller based on the demo's mentioned in the repo - https://github.com/ros-controls/gz_ros2_control/tree/master.

The controller is used in non chained mode. I have setup the plugin to remap tf_odometry messages as follows:
`<remapping>/ackermann_steering_controller/tf_odometry:=/tf</remapping>`

In addition to the controller, the joint state braodcaster also publisher messages to /tf topic. 

When viewed in rviz2, everything works when things are stationary. When the car is moved linearly by publishing twist messages as below with only a linear velocity, the Rviz properly shows the movement is it is in sync with what is shown in gazebo. 

`ros2 topic pub /ackermann_steering_controller/reference_unstamped geometry_msgs/msg/Twist '{linear: {x: 1.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}'`

If I include a steering angle component in the twist message, the movement is not in sync with gazebo. When these is only a steering angle component as below, in Rviz, the car rotates.

`ros2 topic pub /ackermann_steering_controller/reference_unstamped geometry_msgs/msg/Twist '{linear: {x: 0.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.1}}'`

Please see below for a video. Also, when inspected /tf topic with ros2 topic echo, I can see that odom to base_link transform changes continuously. This agrees with what is shown in RViz. 

Since what is set is steering angle and the vehicle movement is zero, can there be a continuous change in odom to base_link?

[Screencast from 07-27-2024 06:23:35 PM.webm](https://github.com/user-attachments/assets/4c4e56db-55d7-4388-854b-c67006c78ec4)

christophfroehlich, 2024-08-07:
Can you share your controller parameters? Are you using open_loop [mode](https://control.ros.org/rolling/doc/ros2_controllers/steering_controllers_library/doc/userdoc.html#parameters)? Could you change that parameter and see if the behavior changes?

christophfroehlich, 2024-08-07:
> In addition to the controller, the joint state braodcaster also publisher messages to /tf topic.

you mean the robot state publisher?

luxapana, 2024-08-08:
The version is 2.35.0.
I tried open loop control, but still its the same. However rotation now is much slower. Please see below video in which I sent only a rotational element as follows:

`ros2 topic pub /ackermann_steering_controller/reference_unstamped geometry_msgs/msg/Twist '{linear: {x: -0.00, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.2}}'`

[Screencast from 08-08-2024 06:45:06 AM.webm](https://github.com/user-attachments/assets/2a05ab78-7fd0-43a5-a40d-b7afbcee1bfe)

christophfroehlich, 2024-08-08:
Would you like to share your robot, or maybe even contribute them to gz_ros2_control_demos?

But actually, this should be reproducible with Mock Hardware as well if you have the same behavior with open loop control.