ros-controls/gz_ros2_control #377: Missing Derivative Gain for Position Command

Tiz000, 2024-07-29:
Currently, the `gz_ros2_control` package implements only a proportional gain control for position commands. 
Is there a specific reason why only proportional gain is implemented without derivative gain in the position control strategy?

Tiz000, 2024-07-29:
I understand that, but I don't want to add my control tasks into the plugin, I would like to implement a better way in which position commands are sent to the joints to simulate the internal dynamic of my motors and make the simulation more realistic. Thanks for your fast response.

christophfroehlich, 2024-07-29:
> I would like to implement a better way in which position commands are sent to the joints to simulate the internal dynamic of my motors and make the simulation more realistic.

That's exactly what the linked PR wants to achieve. Contributions are very welcome :)

Tiz000, 2024-07-31:
Thank you so much

closed 2024-07-31 by Tiz000