ros-controls/gz_ros2_control #389: Unable to build on ros2 humble

siddharth-bhargav, 2024-08-01:
Hi, I was trying to follow the [first Moveit 2 tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) and get it setup, but I was having issues when running `colcon build --mixin release`

I keep getting the following error: 
```
--- stderr: gz_ros2_control
CMake Error at CMakeLists.txt:27 (find_package):
  By not providing "Findgz_sim_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz_sim_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "gz_sim_vendor"
  with any of the following names:

    gz_sim_vendorConfig.cmake
    gz_sim_vendor-config.cmake

  Add the installation prefix of "gz_sim_vendor" to CMAKE_PREFIX_PATH or set
  "gz_sim_vendor_DIR" to a directory containing one of the above files.  If
  "gz_sim_vendor" provides a separate development package or SDK, be sure it
  has been installed.
```
I tried following the solutions suggested through [this thread](https://github.com/doosan-robotics/doosan-robot2/issues/61) and made sure that I am running a compatible version of Gazebo with humble (Fortress). It still isn't working. 

I also tried cloning a copy of the gz_sim_vendor repo into my workspace, which did work, but then I got the following error:

```
--- stderr: gz_sim_vendor                                             
CMake Error at CMakeLists.txt:22 (find_package):
  By not providing "Findgz_cmake_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz_cmake_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "gz_cmake_vendor"
  with any of the following names:

    gz_cmake_vendorConfig.cmake
    gz_cmake_vendor-config.cmake

  Add the installation prefix of "gz_cmake_vendor" to CMAKE_PREFIX_PATH or
  set "gz_cmake_vendor_DIR" to a directory containing one of the above files.
  If "gz_cmake_vendor" provides a separate development package or SDK, be
  sure it has been installed.

```
I don't want to have to do this for ever repository because that seems tedious and like I'm not actually solving the problem. What might I be going wrong?

christophfroehlich, 2024-08-01:
See the table in the README: please use the `humble` branch.

siddharth-bhargav, 2024-08-02:
Thanks! Your comment helped me realize that I was using the wrong branch in MoveIt as well.

closed 2024-08-02 by siddharth-bhargav