ros-controls/gz_ros2_control #390: [gz_ros2_control-system] Could not find shared library

Juancams, 2024-08-09:
Hi!

I'm trying to get the tiago robot working inside harmonic with ros 2 jazzy and I'm having the following problem:

```bash
[gz-1] [Err] [SystemLoader.cc:92] Failed to load system plugin [gz_ros2_control-system] : Could not find shared library.
```

This is what I have in my ros2_control.urdf.xacro

```xml
...

<!-- Init the ROS 2 control gazebo interface -->
    <xacro:if value="${use_sim_time}">
      <gazebo>
        <plugin name="gz_ros2_control::GazeboSimROS2ControlPlugin" filename="gz_ros2_control-system">
          <parameters>$(find tiago_description)/ros2_control/gazebo_controller_manager_cfg.yaml</parameters>
        </plugin>
      </gazebo>
    </xacro:if>

    <!-- ROS 2 control config-->
    <ros2_control name="ros2_control_tiago_system" type="system">

      <!-- Execution env config -->
      <hardware>
        <xacro:if value="${use_sim_time}">
          <plugin>gz_ros2_control/GazeboSimSystem</plugin>

        <!-- IMU sensor (only in simulation)-->
        <xacro:ros2_control_imu name="base_imu_sensor"/>
        </xacro:if>
        <xacro:unless value="${use_sim_time}">
          <plugin>robot_control/RobotControl</plugin>
        </xacro:unless>
      </hardware>

...

```

In another simulator that I have in which I use a UR10 I have no problem putting it like that, but here there is no way and I have tried to call it in different ways.

Thank you so much!

christophfroehlich, 2024-08-11:
First of all, can you run the [gz_ros2_control_demos](https://control.ros.org/jazzy/doc/gz_ros2_control/doc/index.html#gz-ros2-control-demos) successfully?

Juancams, 2024-08-11:
Yes, they work correctly.

christophfroehlich, 2024-08-12:
You should cleanup the part from your snippet, why do you have several `plugin` tags? Use a clean simple working one from the demos, and put some xacro conditions afterwards if you need them.

n4495, 2024-09-25:
I had a similar problem where I could run the gz_ros2_control_demos without issue but when I tried to use my own URDF+launch files GZ threw the "Could not find shared library" error. Even running the diff_drive_example.launch.py from the examples with my launch script gave the same result. So I assume the problem is in my launch file.

Turns out the shell var `GZ_SIM_SYSTEM_PLUGIN_PATH` wasn't set so I pointed it to where I found libgz_ros2_control-system.so.

`export GZ_SIM_SYSTEM_PLUGIN_PATH=/opt/ros/jazzy/lib/`

This worked but I'm still not sure why the example launch has this set and mine does not.

Amronos, 2024-11-01:
I also had a similar problem. In my case I think it was caused by using the [gz_server.launch.py](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim/launch/gz_server.launch.py) launch file. The launch file uses [gzserver.cpp](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim/src/gzserver.cpp) to just start a gz server. I don't encounter this problem when using the [gz_sim.launch.py](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim/launch/gz_sim.launch.py.in) launch file which the demos use.

siyengar04, 2024-11-12:
> I had a similar problem where I could run the gz_ros2_control_demos without issue but when I tried to use my own URDF+launch files GZ threw the "Could not find shared library" error. Even Using the URDF from the examples with my launch script gave the same result. So I assume the problem is in my launch file.
> 
> Turns out the shell var `GZ_SIM_SYSTEM_PLUGIN_PATH` wasn't set so I pointed it to where I found libgz_ros2_control-system.so.
> 
> `export GZ_SIM_SYSTEM_PLUGIN_PATH=/opt/ros/jazzy/lib/`
> 
> This worked but I'm still not sure why the example launch has this set and mine does not.

can confirm, this workaround seems to work for me.

Amronos, 2025-01-28:
This issue can be closed now as https://github.com/gazebosim/ros_gz/pull/659 fixes it.

closed 2025-01-28 by saikishor