ros-controls/gz_ros2_control #391: [Jazzy] How to set: position_proportional_gain

firesurfer, 2024-08-16:
Hi I am looking for a way to set the `position_proportional_gain` parameter. 

According to: https://github.com/ros-controls/gz_ros2_control/pull/250 it should be possible to be set from a parameter file.
Unfortunately there seems to be no (or I can't find it) documentation on how to provide an additional configuration file for the `gz_ros_control` node from a launch file.

firesurfer, 2024-08-22:
A friendly push :)

christophfroehlich, 2024-08-22:
I neither was able to pass ROS arguments to the gz_ros2_control node, I added a parser from the URDF in the linked PR.

christophfroehlich, 2024-08-22:
I managed to pass ROS parameters: be aware that the node is called gz_ros_control instead of gz_ros2_control ;) 
```yaml
gz_ros_control:
  ros__parameters:
    position_proportional_gain: 0.5
```

firesurfer, 2024-08-23:
Sorry for the dumb question, but in a launch file. Which component do I need to pass that parameter file, is it the
```
	start_gazebo_cmd = IncludeLaunchDescription(
		PythonLaunchDescriptionSource(
			os.path.join(pkg_ros_gz_sim, 'launch', 'gz_sim.launch.py')),
		launch_arguments=[('gz_args', [' -r -v 4 empty.sdf']),])
```
 As far as I am aware there is no example that shows how to pass a parameter file.

christophfroehlich, 2024-08-23:
I tried the same, and parameters should have wildcard namespace from the launch file -> it did not work.
I have no idea if this is possible, maybe the gz_sim.launch.py does not pass the parameters to gz.

firesurfer, 2024-08-26:
> I managed to pass ROS parameters: Just add this to the yaml file passed in the `parameters`
> 
> ```yaml
> gz_ros_control:
>   ros__parameters:
>     position_proportional_gain: 0.5
> ```
> 
> be aware that the node is called gz_ros_control instead of gz_ros2_control ;)

Then I don't understand your answer :D 

Or is the PR you linked necessary for that to work ?

christophfroehlich, 2024-08-26:
No, the PR is for adding them as parameters in the URDF plugin tag.

Add gz_ros2_control namespace in the yaml file you load into the plugin, like this cart_controller.yaml
```
<gazebo>
  <plugin filename="libgz_ros2_control-system.so" name="gz_ros2_control::GazeboSimROS2ControlPlugin">
    <parameters>$(find gz_ros2_control_demos)/config/cart_controller.yaml</parameters>
  </plugin>
</gazebo>
```
They will be passed to the node of gz_ros_control.

PR #393 (2024-08-26): Parse `position_proportional_gain` parameter from URDF and update docs
Now it can be also set via the plugin tag in the URDF.
Fixes #391
PR #410 (2024-08-26): Parse `position_proportional_gain` parameter from URDF and update docs (backport #393)
Now it can be also set via the plugin tag in the URDF.
Fixes #391 <hr>This is an automatic backport of pull request #393 done by [Mergify](https://mergify.com).
closed 2024-08-26 by ahcorde