ros-controls/gz_ros2_control #394: [HELP]: How to build binaries for gz_ros2_control

Witty-Wizard, 2024-08-23:
## System Config:
ROS Distro : Humble
Gazebo Version: Harmonic
Linux Version: Ubuntu Jammy (Running Inside Docker)

## Help Needed:
I need help on how to install and setup `gz_ros2_control plugin` on `ROS Humble` with `Gazebo Harmonic`.

While Compiling from source I get the following error:
```bash
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
gz_ros2_control: Cannot locate rosdep definition for [gz_plugin_vendor]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

christophfroehlich, 2024-08-23:
Have you read the compatibility table in the [readme](https://github.com/ros-controls/gz_ros2_control?tab=readme-ov-file#gz_ros2_control)? This combination is not officially supported.
For humble distro you have to use the `humble` branch, because the `*_vendor` packages are not released for distros older than jazzy. But this might not work with harmonic, you can try to use the iron branch.

Witty-Wizard, 2024-08-23:
> Have you read the compatibility table in the [readme](https://github.com/ros-controls/gz_ros2_control?tab=readme-ov-file#gz_ros2_control)? This combination is not officially supported. For humble [...]

I have read the documentation, but I want to run `harmonic` with `humble`.

christophfroehlich, 2024-08-23:
Have you tried humble or iron branch as suggested?

Witty-Wizard, 2024-08-23:
I did try those branches, there is get dependency error

christophfroehlich, 2024-08-23:
Sorry, this is just not supported. What you can try is to compile the iron version of the whole ros2_control stack, this could work on humble distro but without guarantee to be stable.

Witty-Wizard, 2024-08-24:
I tried building using the `iron` branch, I got the Following error:
```bash
Starting >>> gz_ros2_control
--- stderr: gz_ros2_control                             
/home/jazzer/gz_ros2_control_ws/src/gz_ros2_control/gz_ros2_control/src/gz_ros2_control_plugin.cpp: In member function ‘virtual void gz_ros2_control::GazeboSimROS2ControlPlugin::Configure(const Entity&, const std::shared_ptr<const sdf::v14::Element>&, gz::sim::v8::EntityComponentManager&, gz::sim::v8::EventManager&)’:
/home/jazzer/gz_ros2_control_ws/src/gz_ros2_control/gz_ros2_control/src/gz_ros2_control_plugin.cpp:428:67: error: ‘__gnu_cxx::__alloc_traits<std::allocator<hardware_interface::HardwareInfo>, hardware_interface::HardwareInfo>::value_type’ {aka ‘struct hardware_interface::HardwareInfo’} has no member named ‘hardware_plugin_name’
  428 |     std::string robot_hw_sim_type_str_ = control_hardware_info[i].hardware_plugin_name;
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/gz_ros2_control-system.dir/build.make:76: CMakeFiles/gz_ros2_control-system.dir/src/gz_ros2_control_plugin.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/gz_ros2_control-system.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
```

This might be fixable with some tweaking in the source code.
I will try and update it here.

Witty-Wizard, 2024-08-24:
> I tried building using the `iron` branch, I got the Following error:
> 
> ```shell
> Starting >>> gz_ros2_control
> --- stderr: gz_ros2_control                             
> /home/jazzer/gz_ros2_control_ws/src/gz_ros2_control/gz_ros2_control/src/gz_ros2_control_plugin.cpp: In member function ‘virtual void gz_ros2_control::GazeboSimROS2ControlPlugin::Configure(const Entity&, const std::shared_ptr<const sdf::v14::Element>&, gz::sim::v8::EntityComponentManager&, gz::sim::v8::EventManager&)’:
> /home/jazzer/gz_ros2_control_ws/src/gz_ros2_control/gz_ros2_control/src/gz_ros2_control_plugin.cpp:428:67: error: ‘__gnu_cxx::__alloc_traits<std::allocator<hardware_interface::HardwareInfo>, hardware_interface::HardwareInfo>::value_type’ {aka ‘struct hardware_interface::HardwareInfo’} has no member named ‘hardware_plugin_name’
>   428 |     std::string robot_hw_sim_type_str_ = control_hardware_info[i].hardware_plugin_name;
>       |                                                                   ^~~~~~~~~~~~~~~~~~~~
> gmake[2]: *** [CMakeFiles/gz_ros2_control-system.dir/build.make:76: CMakeFiles/gz_ros2_control-system.dir/src/gz_ros2_control_plugin.cpp.o] Error 1
> gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/gz_ros2_control-system.dir/all] Error 2
> gmake[1]: *** Waiting for unfinished jobs....
> gmake: *** [Makefile:146: all] Error 2
> ```
> 
> This might be fixable with some tweaking in the source code. I will try and update it here.

I Did It!!!

changing `hardware_plugin_name` to `hardware_class_type` in the `gz_ros2_control_plugin.cpp` file did the trick.
This works because `hardware_plugin_name` has been changed to `hardware_class_type` for `ros humble`.

I will create a new branch in my fork called `humble-harmonic`.

christophfroehlich, 2024-08-24:
fyi, there is a similar fork for humble+garden: https://github.com/ros-controls/gz_ros2_control/pull/268

It might have been possible to also compile iron version of ros2_control+controllers from source instead of changing the variable, for example with [this repos file](https://github.com/ros-controls/ros2_control_ci/blob/master/ros_controls.iron.repos).

Witty-Wizard, 2024-08-24:
> fyi, there is a similar fork for humble+garden: #268
>
> It might have been possible to also compile iron version of ros2_control+controllers from source instead of changing the variable, for example [...]

That is the idea, I am trying to understand the CI process so that I can publish a branch, and make it available through `apt`, this way I can use it in my `docker image`.

christophfroehlich, 2024-08-26:
THB I don't think that this is worth it.
But if you want to fork this repo you could release it as independent package on the ROS buildfarm.  Or you use github to build the debian and release it.

danisotelo, 2024-11-02:
Hi! Did you finally made it available through apt? I am also interested in using it in a docker image.

Witty-Wizard, 2024-11-02:
> Hi! Did you finally made it available through apt? I am also interested in using it in a docker image.

I ended up switching to `gazebo-ignition`. As I had a deadline to meet, However, Now seems like a good time to revisit this.

https://github.com/Robotics-Society-PEC/Robotic-Arm

danisotelo, 2024-11-02:
Thank you! Mmm, yes I think I'll continue to try to get it to work with Harmonic because I need to use a robotic arm together with a drone using PX4. Any advice?

danisotelo, 2024-11-02:
Perhaps, the easiest way would be to update to Ubuntu 24 and use ROS2 Jazzy?

Witty-Wizard, 2024-11-02:
> Perhaps, the easiest way would be to update to Ubuntu 24 and use ROS2 Jazzy?

I did try that but didn't have any luck getting moveit2 working with Jazzy, thats why I used humble.

You can try porting my repo to jazzy or I might create a nee branch to port this to Jazzy

danisotelo, 2024-11-02:
Thank you very much!