ros-controls/gz_ros2_control #440: ros2 controllers are not working with bullet-featherstone plugin in gz harmonic

vamsikalagaturu, 2024-11-22:
## Environment
* OS Version: Ubuntu 24.04 
* binary build 
   - ros-jazzy-gz-ros2-control-demos/noble,now 1.2.8-1noble.20241118.194255 amd64 [installed]
   - ros-jazzy-gz-ros2-control/noble,now 1.2.8-1noble.20241118.193506 amd64 [installed]

* gazebo harmonic (binary build)
   - ros-jazzy-ros-gz-sim/noble,now 1.0.7-1noble.20241115.212712 amd64 [installed,automatic]

## Description

When the bullet-featherstone physics engine is used with gz harmonic, the controllers are not getting into active state and are getting timed out. sometimes, even if they are transitioned into active state, I am unable to control the joints.  When the plugin is changed to default or  dartsim, the controllers are being loaded and executing as expected (however, they dont support mimic joints as we are aware)

## Steps to reproduce

1. ros2 launch gz_ros2_control_demos gripper_mimic_joint_example_position.launch.py
2. ros2 run gz_ros2_control_demos example_gripper
3. you can observe that the gripper joints are not moving. 

This is happening not just with the demo, but also with the custom packages that I am working with.

vamsikalagaturu, 2024-11-30:
Is there any update on this?

MarqRazz, 2025-03-10:
I just tested 
`ros2 launch gz_ros2_control_demos gripper_mimic_joint_example_position.launch.py` with 
`ros2 run gz_ros2_control_demos example_gripper ` and it works as expected.

It also works as expected with my [custom robot](https://github.com/MarqRazz/piper_ros2/blob/main/piper_bringup/worlds/empty_world.sdf#L14-L17) and [gripper](https://github.com/MarqRazz/piper_ros2/blob/main/piper_description/urdf/piper_gripper_macro.xacro).

vamsikalagaturu, 2025-04-15:
Thanks for the confirmation. The example is working now on my system too. There are new changes anyway since I created the issue.  Closing the issue now.

closed 2025-04-15 by vamsikalagaturu