ros-controls/gz_ros2_control #456: /robot_controller/joint_trajectory topic is not translated to Gazebo Sim

julianhfmn, 2024-12-19:
Hello,

I am on ROS2 Jazzy, Moveit2 and Gazebo Sim Harmonic. I use a custom robot 6DoF Robot which I did the moveit2 setup assistant for, and using the robot with moveit2 in Gazebo Sim Harmonic works. 

My issue is, that although launching moveit servo with my custom robot without gazebo is functional, I cannot use it combined with Gazebo Sim.

The issue is see is the Gazebo Plugin for ROS2 Control. 

When launching the robot without gazebo, I can use the hardware plugin in my ros2_control.xacro file as this:

        <hardware>
            <plugin>mock_components/GenericSystem</plugin>
        </hardware>

When launching my robot in gazebo, I use the gz_ros2_control package as this: 

        <hardware>
            <plugin>gz_ros2_control/GazeboSimSystem</plugin>
        </hardware>

In the case of launching both gazebo and moveit servo, and only launching mock_components plugin, the robot controller I define in gazebo cannot be activated:

    [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
The robot is not actuated and flops onto the ground. This makes sense, as the gazebo plugin is missing.

In the case of launching `gz_ros2_control/GazeboSimSystem` as my hardware plugin, I can see that my `/robot_controller/joint_trajectory` topic is populated, but neither the robot in rviz2 nor in gazebo move. Somehow, the `gz_ros2_control/GazeboSimSystem` does not pick up this topic, although it should? I can see, that the `/robot_controller` node is subscribed to `/robot_controller/joint_trajectory`.

I tested this with ompl and the standard moveit2 p2p motion I can set in the MotionPlanning plugin in rviz, and the `/robot_controller/joint_trajectory` is empty there, too.

How can I get the gazebo sim plugin to transfer my joint trajectory to gazebo and move the simulated robot?
Is it another topic I need to publish to for the plugin to work?

Thanks a lot in advance!

All the best and merry christmas,

Julian

julianhfmn, 2024-12-19:
I made some new findings: 
Using either joint or position controllers with a `std_msgs/msg/Float64MultiArray` type is functional in gazebo.
It seems that only the /robot_controller/joint_trajectory topic is not working?

christophfroehlich, 2024-12-21:
Can you only launch gazebo with the robot and its controller and test it with the [rqt_joint_trajectory_controller](https://control.ros.org/rolling/doc/ros2_controllers/rqt_joint_trajectory_controller/doc/userdoc.html)? It uses the same topic.

julianhfmn, 2024-12-22:
I tested what you proposed and neither in gazebo nor in rviz2 the joints move. So I assume even with a correct joint_trajectory message, somehow the control signals don't reach the sim.

christophfroehlich, 2024-12-22:
can you provide an example? is it a public repo?

christophfroehlich, 2024-12-22:
Furthermore, cart_example_position uses a joint_trajectory controller and this works as it should with rqt_joint_trajectory_controller:
`ros2 launch gz_ros2_control_demos cart_example_position.launch.py`

tested on rolling with gz_ros2_control from source
```
ros-rolling-ros-gz-bridge/noble,now 2.1.2-1noble.20241204.195601 amd64 [installed]
ros-rolling-ros-gz-interfaces/noble,now 2.1.2-1noble.20241125.160511 amd64 [installed,automatic]
ros-rolling-ros-gz-sim/noble,now 2.1.2-1noble.20241205.120818 amd64 [installed]
```
Can you reproduce that with jazzy/harmonic?

phantson, 2024-12-24:
Same issue here, using Gazebo and Ros Jazzy i have this error. Furthermore I can't even launch cart_example_position.launch.py, as it closes Gazebo instantly returning this error : GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v145WorldE], which doesn't have `operator>>`. Component will not be deserialized.
[gazebo-1] [GUI] [Msg] Loaded plugin [EntityTree] from path [/opt/ros/jazzy/opt/gz_sim_vendor/lib/gz-sim-8/plugins/gui/libEntityTree.so]

christophfroehlich, 2024-12-24:
> Same issue here, using Gazebo and Ros Jazzy with MoveIt2 i have this error. Furthermore I can't even launch cart_example_position.launch.py, as it closes Gazebo instantly returning this error : 
> 
> ```
> [GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v145WorldE], which doesn't have `operator>>`. Component will not be deserialized.
> [gazebo-1] [GUI] [Msg] Loaded plugin [EntityTree] from path [/opt/ros/jazzy/opt/gz_sim_vendor/lib/gz-sim-8/plugins/gui/libEntityTree.so]
> ```
> 

This looks like an ABI break or incompatible gz version. How have you installed gz and gz_ros2_control? Have you upgraded anything individually instead of `sudo apt-get upgrade` everything?

julianhfmn, 2024-12-24:
> Can you reproduce that with jazzy/harmonic?

I tested it on jazzy. Only the `ros2 run gz_ros2_control_demos example_position` script works for me, for the effort and velocity controllers I only receive a:
`[INFO] [1735055303.141716459] [effort_test_node]: node created
`
and the code stops again, without moving the robot.

Interestingly, although the robot moves in gazebo, the `/joint_trajectory_controller/joint_trajector` ros2 topic is not populated? I echo the topic and receive nothing, is it the same for you @christophfroehlich?

gz_ros2_control as well as the demos are installed as binaries for me. 

My own repo sadly is not public, but using the https[://moveit.picknik.ai/main/doc/examples/realtime_servo/realtime_servo_tutorial.html
](moveit servo tutorial)  with the panda:
`ros2 launch moveit_servo demo_ros_api.launch.py` 
I can move the robot and I also can echo the joint_trajectory topic, which works in my repo as well, only that the results are not translated to gazebo.
I could try a source install and see if that changes anything?
Although right now the velocity control works for me, too.

christophfroehlich, 2024-12-24:
> I tested it on jazzy. Only the `ros2 run gz_ros2_control_demos example_position` script works for me, for the effort and velocity controllers I only receive a:
> `[INFO] [1735055303.141716459] [...]

the example_ executables create a publisher and send some values, but don't launch gazebo. Try the launch files instead. See the [documentation](https://control.ros.org/jazzy/doc/gz_ros2_control/doc/index.html#cart-on-rail).

julianhfmn, 2025-01-14:
> the example_ executables create a publisher and send some values, but don't launch gazebo. Try the launch files instead. See the [documentation](https://control.ros.org/jazzy/doc/gz_ros2_control/doc/index.html#cart-on-rail).

Ah my fault, yes the trajectory controller does work when using the launch file. `ros2 launch gz_ros2_control_demos cart_example_position.launch.py`.

I also now found out that the trajectory controller also works when creating a moveit_config of the robot and launching demo.launch.py - but not inside gazebo.

Currently, I assume I have something wrong in my launch or config files, so I'll dig a bit deeper.

julianhfmn, 2025-01-14:
Okay, I think my issue is solved. 

I have multiple controllers I have loaded, one JointTrajectoryController for a Linear Axis + Robot, and two velocity controllers for the linear axis, one for my robot. 

I had all of them in the active state.

Setting the velocity controllers to inactive, fixed the JointTrajectoryController such that I can use it with rqt, or a move group in gazebo harmonic.

I am not sure if that is intended behavior and some sort of warning would have accelerated my search, but that's how I fixed my issue now.

Thanks for the help @christophfroehlich

closed 2025-01-14 by julianhfmn

julianhfmn, 2025-01-14:
Right now, I have: 
`$ ros2 control list_controllers `
```
lifted_arm_controller joint_trajectory_controller/JointTrajectoryController  active  
lift_controller       velocity_controllers/JointGroupVelocityController      inactive
arm_controller        velocity_controllers/JointGroupVelocityController      inactive
joint_state_broadcaster     joint_state_broadcaster/JointStateBroadcaster          active  
```
and:

`$ ros2 control list_hardware_interfaces`
```
command interfaces
	Revolute_10/position [available] [claimed]
	Revolute_10/velocity [available] [unclaimed]
	Revolute_12/position [available] [claimed]
	Revolute_12/velocity [available] [unclaimed]
	Revolute_14/position [available] [claimed]
	Revolute_14/velocity [available] [unclaimed]
	Revolute_17/position [available] [claimed]
	Revolute_17/velocity [available] [unclaimed]
	Revolute_6/position [available] [claimed]
	Revolute_6/velocity [available] [unclaimed]
	Revolute_8/position [available] [claimed]
	Revolute_8/velocity [available] [unclaimed]
	Slider_18/position [available] [claimed]
	Slider_18/velocity [available] [unclaimed]
state interfaces
	Revolute_10/effort
	Revolute_10/position
	Revolute_10/velocity
	Revolute_12/effort
	Revolute_12/position
	Revolute_12/velocity
	Revolute_14/effort
	Revolute_14/position
	Revolute_14/velocity
	Revolute_17/effort
	Revolute_17/position
	Revolute_17/velocity
	Revolute_6/effort
	Revolute_6/position
	Revolute_6/velocity
	Revolute_8/effort
	Revolute_8/position
	Revolute_8/velocity
	Rigid_32/position
	Rigid_33/position
	Slider_18/effort
	Slider_18/position
	Slider_18/velocity
```

Nevertheless, I am able to set the other controllers active, so that:
```
$ ros2 control list_controllers 
[INFO] [1736873930.251561591] [_ros2cli_11661]: waiting for service /controller_manager/list_controllers to become available...
aiina_lifted_arm_controller joint_trajectory_controller/JointTrajectoryController  active
aiina_lift_controller       velocity_controllers/JointGroupVelocityController      active
aiina_arm_controller        velocity_controllers/JointGroupVelocityController      active
joint_state_broadcaster     joint_state_broadcaster/JointStateBroadcaster          active
```
As far as I see it, it should be only the velocity interface which is then claimed. Still, somehow I cannot use move_group for example to send a JointTrajectoryCommand, when the velocity controllers are active. The robot does not move in rviz2 nor in gazebo, but the node tells me: `[move_group-1] [INFO] [1736874047.943644600] [move_group.moveit.moveit.ros.move_group.move_action]: Solution was found and executed.`

- When I start the controllers, but the velocity controllers are inactive, I can send commands to the robot to move via move_group.
- When I then activate the velocity controllers again, I cannot move the robot via move_group.

christophfroehlich, 2025-01-14:
Do I understand that correctly, that the joint_trajectory_controller claims position interface, but the others claim the velocity? Afaik this is a valid configuration, but not supported/prevented by gz_ros2_control plugin.

julianhfmn, 2025-01-15:
Yes. So far the joint trajectory controller works fine though, I did not see a difference for reaching certain poses at least - I have not tested accuracy. I plan to use moveit servo for my Cartesian paths, which should be accurate then.

Would the recommended approach be to let the joint_trajectory_controller control position and velocity? When deactivating the velocity controllers I guess that would be possible. I probably deactivated the interfaces in the setup assistant when testing or by accident.

christophfroehlich, 2025-01-15:
gazebo does not support joints with position+velocity interfaces, but if they are consistent it "just" works. Problem is when your velocity controller is claiming the interface, then the position command of JTC will be overridden with zeros from the inactive velocity controllers. 
The hardware component can't detect which controller is claiming what, which means that we can't throw a warning in a setup like yours.

julianhfmn, 2025-01-15:
Okay, good to know, thanks! With velocity controllers deactivated, the robot moves with a joint_trajectory as well, so I assume that was the issue in the beginning.