ros-controls/gz_ros2_control #466: No rolling binaries on build farm

christophfroehlich, 2024-12-26:
https://build.ros2.org/job/Rsrc_uN__gz_ros2_control__ubuntu_noble__source/ fails

moriarty, 2025-01-04:
@ahcorde looks like this job started failing around the same time you ran bloom:

https://build.ros2.org/job/Rsrc_uN__gz_ros2_control__ubuntu_noble__source/119/
```
# BEGIN SUBSECTION: get sources
Invoking 'git clone --branch debian/ros-rolling-gz-ros2-control_2.0.3-1_noble --depth 1 --no-single-branch https://github.com/ros2-gbp/ign_ros2_control-release.git /tmp/sourcedeb/source'
Cloning into '/tmp/sourcedeb/source'...
fatal: Remote branch debian/ros-rolling-gz-ros2-control_2.0.3-1_noble not found in upstream origin
```

This error is correct, the branch ending in `_noble` does not exist.
There is tag https://github.com/ros2-gbp/ign_ros2_control-release/releases/tag/release%2Frolling%2Fgz_ros2_control%2F2.0.3-1

https://github.com/ros2-gbp/ign_ros2_control-release
```
gz_ros2_control (rolling) - 2.0.3-1
The packages in the gz_ros2_control repository were released into the rolling distro by running /usr/bin/bloom-release --rosdistro rolling --track rolling gz_ros2_control on Wed, 11 Dec 2024 15:32:24 -0000
```

ahcorde, 2025-01-20:
@bmagyar did a new release https://github.com/ros-controls/gz_ros2_control/commits/rolling/

christophfroehlich, 2025-01-20:
yes but he was not able to release it to [rosdistro](https://github.com/ros/rosdistro/pulls?q=is%3Apr+gz_ros2_control+is%3Aclosed), and the[ build job is disabled](https://build.ros2.org/job/Rsrc_uN__gz_ros2_control__ubuntu_noble__source/)?

christophfroehlich, 2025-02-04:
2.0.5 is built now and available on testing: http://packages.ros.org/ros2-testing/ubuntu/dists/noble/main/binary-amd64/Packages

closed 2025-02-04 by christophfroehlich