ros-controls/gz_ros2_control #467: Plugin fails to load on ROS Jazzy

phantson, 2024-12-26:
While trying to link MoveIt2 and Gazebo in order to simulate my robot with this configuration : 

Ros2 control : 
`<xacro:macro name="niryo_ned2_ros2_control" params="name initial_positions_file">
<xacro:property name="initial_positions" value="${xacro.load_yaml(initial_positions_file)['initial_positions']}"/>
<ros2_control name="${name}" type="system">
<hardware>

<plugin>mock_components/GenericSystem</plugin>
</hardware>
<joint name="joint_1">
<command_interface name="position"/>
<state_interface name="position">
<param name="initial_value">${initial_positions['joint_1']}</param>
</state_interface>
</joint>
<joint name="joint_2">
<command_interface name="position"/>
<state_interface name="position">
<param name="initial_value">${initial_positions['joint_2']}</param>
</state_interface>
</joint>
<joint name="joint_3">
<command_interface name="position"/>
<state_interface name="position">
<param name="initial_value">${initial_positions['joint_3']}</param>
</state_interface>
</joint>
<joint name="joint_4">
<command_interface name="position"/>
<state_interface name="position">
<param name="initial_value">${initial_positions['joint_4']}</param>
</state_interface>
</joint>
<joint name="joint_5">
<command_interface name="position"/>
<state_interface name="position">
<param name="initial_value">${initial_positions['joint_5']}</param>
</state_interface>
</joint>
<joint name="joint_6">
<command_interface name="position"/>
<state_interface name="position">
<param name="initial_value">${initial_positions['joint_6']}</param>
</state_interface>
</joint>
</ros2_control>
</xacro:macro>`

Ros2-gz : 
`<gazebo>
<plugin filename="libgz_ros2_control-system.so" name="gz_ros2_control::GazeboSimROS2ControlPlugin">
<parameters>
/home/phantson/ws_simuxx/niryo_robot_ned2_config/config/ros2_controllers.yaml
</parameters>
</plugin>
</gazebo>`

The plugin fails to load returning this error : 
`[ERROR] [1735203845.630924770] [gz_ros_control.GZResourceManager]: The plugin failed to load for some reason. Error: According to the loaded plugin descriptions the class mock_components/GenericSystem with base class type gz_ros2_control::GazeboSimSystemInterface does not exist. Declared types are  gz_ros2_control/GazeboSimSystem ign_ros2_control/IgnitionSystem
`

The robot actually shows up on Gazebo but it is not responsive to MoveIt while asking to execute a trajectory. (Even if /joint_state is correctly working)

christophfroehlich, 2024-12-26:
why do you use [mock_components/GenericSystem](https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/mock_components_userdoc.html) if you use gazebo?
See the gz_ros2_control_demos how this typically is set up, e.g. [here](https://github.com/ros-controls/gz_ros2_control/blob/rolling/gz_ros2_control_demos/urdf/test_diff_drive.xacro.urdf#L149-L179).

christophfroehlich, 2025-01-29:
Closing due to inactivity for a month.

closed 2025-01-29 by christophfroehlich