ros-controls/gz_ros2_control #469: Velocity command interface is broken on rolling+ionic

christophfroehlich, 2024-12-28:
**Describe the bug**
Velocity command interface is broken on rolling+ionic? position interfaces work

**To Reproduce**
Steps to reproduce the behavior:
1. `ros2 launch gz_ros2_control_demos diff_drive_example.launch.py`
2. `ros2 run gz_ros2_control_demos example_diff_drive `
3. nothing moves

same with cart_velocity example

**Environment (please complete the following information):**
 - OS: Ubuntu noble
 - Version rolling from source, ros-rolling-gz-sim-vendor 0.2.0

**Additional context**
The same examples work on humble+fortress.

saikishor, 2024-12-29:
I've tested this on Jazzy and it is working. I will also test it on rolling and update here

christophfroehlich, 2024-12-30:
I can confirm, it works with jazzy and harmonic 8.7. I just have rebuilt my rolling docker and workspace, but still the robots don't move.

saikishor, 2025-01-01:
I confirm that in rolling it doesn't work 🙏🏽

christophfroehlich, 2025-01-01:
@azeey Could you please help us here? What has changed from Harmonic to Ionic, so that velocity commands are not working any more?

JuuHuu, 2025-01-02:
After the update, my differential car cannot move in Gazebo Harmonic + Jazzy. However it can be controlled before the update. Finally i find it is because i set `linear.x.max_jerk:` and other jerk limits in controller config file  to `0`. It should be set to `.NAN`.

christophfroehlich, 2025-01-02:
If you have had has_jerk_limits=false in your parameter file, it should have printed you a warning? Otherwise it should not have worked before already 🤔 
However, this is not related to this issue here.

azeey, 2025-01-06:
There's a breaking change in Ionic where the `JointVelocityCmd` is deleted after each time step. This was an intentional change and is covered in the [Migration](https://github.com/gazebosim/gz-sim/blob/gz-sim9/Migration.md#gazebo-sim-8x-to-90) document. I'll also add it as a breaking change in https://github.com/gazebosim/gz-ionic/blob/main/highlights.md. Alex's https://github.com/ros-controls/gz_ros2_control/pull/450 should fix the issue.

christophfroehlich, 2025-01-07:
Oops, thanks for the hint to #450. I forgot that in the meantime..

closed 2025-01-07 by ahcorde