ros-controls/gz_ros2_control #474: Issues on connecting ros2_control to gazebo

Elarak, 2025-01-02:
Hello, I'm using ROS2 Jazzy and I have issues on connected ros2 to gazebo.
I want to control the 18 joints of my robot with forward_position_controller. 

I have 2 controllers initialized : ForwardCommandController and JointStateBroadcaster
ros2 control list_controllers
forward_position_controller forward_command_controller/ForwardCommandController  active
joint_state_broadcaster     joint_state_broadcaster/JointStateBroadcaster        active

We can see that all my joints are available
ros2 control list_hardware_interfaces
command interfaces
	joint_patte_1A_patte_1B/position [available] [claimed]
	joint_patte_1B_patte_1C/position [available] [claimed]
	joint_patte_1C_patte_1D/position [available] [claimed]
	joint_patte_2A_patte_2B/position [available] [claimed]
	joint_patte_2B_patte_2C/position [available] [claimed]
	joint_patte_2C_patte_2D/position [available] [claimed]
	joint_patte_3A_patte_3B/position [available] [claimed]
	joint_patte_3B_patte_3C/position [available] [claimed]
	joint_patte_3C_patte_3D/position [available] [claimed]
	joint_patte_4A_patte_4B/position [available] [claimed]
	joint_patte_4B_patte_4C/position [available] [claimed]
	joint_patte_4C_patte_4D/position [available] [claimed]
	joint_patte_5A_patte_5B/position [available] [claimed]
	joint_patte_5B_patte_5C/position [available] [claimed]
	joint_patte_5C_patte_5D/position [available] [claimed]
	joint_patte_6A_patte_6B/position [available] [claimed]
	joint_patte_6B_patte_6C/position [available] [claimed]
	joint_patte_6C_patte_6D/position [available] [claimed]

I can read the initial position of my joints with JointStateBroadcaster

And my publisher is publishing every 1 second between 2  valid positions.
We can see a typical message : 
ros2 topic echo forward_position_controller/commands
layout:
  dim: []
  data_offset: 0
data:
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
---

But on gazebo, the robot isn't moving. I don't have any changes on the joint_state topic. I don't have new messages on the terminal of my launch 
Am I missing something ?

christophfroehlich, 2025-01-02:
@kdeshp-prof you are also using gz_ros2_control on Jazzy, with harmonic, and have a position command interface? 
Can you confirm that `cart_example_position.launch.py` works?

kdeshp-prof, 2025-01-02:
Yes I am also using gz_ros2_control on jazzy with harmonic, but I have a mobile manipulator so I am using differentialDrive Controller for base, JointTrajectoryController for the manipulator joints. The issue is all my controllers are getting active 
[image]
as shown but, the manipulator does not move to init position as defined for each joint. simillarly the differential drive control is completely un-responsive to velocity commands.
Finally, Yes I confirm that the cart_example and differential drive examples both are working fine individually.

saikishor, 2025-01-02:
Hello @kdeshp-prof!

Can you provide us with a minimal example to reproduce the issue? That way it would be easier for us to debug.

Thank you

kdeshp-prof, 2025-01-07:
Please use this forked [repository ](https://github.com/kdeshp-prof/articubot_one/tree/new_gazebo)which includes teleop twist keyboard inside launch_sim.launch.py

christophfroehlich, 2025-02-11:
You have configured the publisher wrong. From jazzy on, you have to use stamped messages for the twist message. Change your launchfile to
```python
    teleop_keyboard = Node(
            package="teleop_twist_keyboard",
            executable="teleop_twist_keyboard",
            prefix="xterm -e",
            parameters=[{'stamped': True}],
            remappings=[('cmd_vel','/diff_cont/cmd_vel')]
```
and remove all the bridges except for the clock message, and your robot moves.

christophfroehlich, 2025-03-19:
Closing because of no updates of the original author of the issue.

closed 2025-03-19 by christophfroehlich