ros-controls/gz_ros2_control #476: Controller manager update_rate is wrongly limited by gazebo's simulation rate

tonynajjar, 2025-01-07:
**Describe the bug**

The controller manager update_rate seems to be wrongly limited by gazebo's simulation rate

**To Reproduce**
Steps to reproduce the behavior:

I have a world with 60Hz update_rate and RTF 1.0 which is reached:
```
    <physics name="1ms" type="ode">
      <max_step_size>0.01666666</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>
```

My controller_manager has an update_rate of 50 Hz:

```
/**:
  controller_manager:
    ros__parameters:
      update_rate: 50 # Hz
```

**Expected behavior**

I expect the controller manager to run at 50Hz, however the diagnostics complain that only 30Hz is reached.

If I change `max_step_size` to 0.01 (100Hz) then the CM is able to reach the 50Hz

**Screenshots**

[image]

**Environment (please complete the following information):**
 - OS: Ubuntu Noble
 - Version Jazzy

**Additional context**
Add any other context about the problem here, especially include any modifications to ros2_control that relate to this issue.

christophfroehlich, 2025-01-08:
Isn't this an issue that your update rates don't have a common integer divider? The controller_manager update is only triggered with the update rate of gz

https://github.com/ros-controls/gz_ros2_control/blob/f9b4e7f72c42cdb1269c879a0dcfa153070a817e/gz_ros2_control/src/gz_ros2_control_plugin.cpp#L514-L526

saikishor, 2025-01-10:
@tonynajjar I believe the max step should be a common divisor factor higher than the CM loop rate to be able to have proper simulation. For instance, a CM update rate of 100 Hz might need a step time of 0.005 or lower by the possible common factor. If not, you might observe some loop skipping.