ros-controls/gz_ros2_control #493: Launching Two UR Arms in Gazebo with Separate Controllers

sl148, 2025-02-01:
# **Launching Two UR Arms in Gazebo with Separate Controllers**

## **Objective**
I want to launch two UR arms in Gazebo with separate controller managers. The expected behavior is that each arm operates independently, managed by its own controller manager.

### **Steps Overview**
1. **Launch Two UR Arms in Gazebo** (Current Focus)
2. **Integrate the MoveIt Interface** (Next Step)
3. **Deploy on Physical Hardware** (Final Step)

### **Specifications**
ROS2 Jazzy
Ubuntu 24

---

## **Launch Files**

ur_sim_control.launch.py

```
from launch import LaunchDescription
from launch.actions import (
    DeclareLaunchArgument,
    IncludeLaunchDescription,
    OpaqueFunction,
    RegisterEventHandler,
)
from launch.conditions import IfCondition, UnlessCondition
from launch.event_handlers import OnProcessExit
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import (
    Command,
    FindExecutable,
    LaunchConfiguration,
    PathJoinSubstitution,
    IfElseSubstitution,
)
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from launch.substitutions import LaunchConfiguration, PythonExpression

def launch_setup(context, *args, **kwargs):
    # Initialize Arguments
    ur_type = LaunchConfiguration("ur_type")
    safety_limits = LaunchConfiguration("safety_limits")
    safety_pos_margin = LaunchConfiguration("safety_pos_margin")
    safety_k_position = LaunchConfiguration("safety_k_position")
    # General arguments
    controller_file = LaunchConfiguration("controller_file")
    tf_prefix = LaunchConfiguration("tf_prefix")
    activate_joint_controller = LaunchConfiguration("activate_joint_controller")
    controller_manager_name = LaunchConfiguration("controller_manager_name")
    initial_joint_controller = LaunchConfiguration("initial_joint_controller")
    description_file = LaunchConfiguration("description_file")
    launch_rviz = LaunchConfiguration("launch_rviz")
    rviz_config_file = LaunchConfiguration("rviz_config_file")
    gazebo_gui = LaunchConfiguration("gazebo_gui")
    world_file = LaunchConfiguration("world_file")
    namespace = LaunchConfiguration("namespace")
    x = LaunchConfiguration("x")
    y = LaunchConfiguration("y")
    z = LaunchConfiguration("z")
    Y = LaunchConfiguration("Y")

    robot_controllers = PathJoinSubstitution(
        [
            FindPackageShare("multi_robot_arm"),
            "config", 'ur', 'ur5e',
            controller_file,
[... 838 lines cut ...]
[spawner-8] [WARN] [1738189684.067779401] [ur2.spawner_joint_state_broadcaster]: Could not contact service /ur2/controller_manager/list_controllers
[spawner-8] [INFO] [1738189684.068945377] [ur2.spawner_joint_state_broadcaster]: waiting for service /ur2/controller_manager/list_controllers to become available...
[spawner-4] [WARN] [1738189684.070417920] [ur1.spawner_scaled_joint_trajectory_controller]: Could not contact service /ur1/controller_manager/list_controllers
[spawner-4] [INFO] [1738189684.071227625] [ur1.spawner_scaled_joint_trajectory_controller]: waiting for service /ur1/controller_manager/list_controllers to become available...
[spawner-3] [WARN] [1738189684.079436654] [ur1.spawner_joint_state_broadcaster]: Could not contact service /ur1/controller_manager/list_controllers
[spawner-3] [INFO] [1738189684.079663861] [ur1.spawner_joint_state_broadcaster]: waiting for service /ur1/controller_manager/list_controllers to become available...
[spawner-9] [WARN] [1738189684.079688990] [ur2.spawner_scaled_joint_trajectory_controller]: Could not contact service /ur2/controller_manager/list_controllers
[spawner-9] [INFO] [1738189684.079863493] [ur2.spawner_scaled_joint_trajectory_controller]: waiting for service /ur2/controller_manager/list_controllers to become available...
[gazebo-11] [WARN] [1738189684.356755867] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[gazebo-11] [WARN] [1738189684.357958367] [gz_ros_control]: Waiting RM to load and initialize hardware...
[ros2_control_node-1] [WARN] [1738189684.805254461] [ur1.controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-6] [WARN] [1738189684.806338577] [ur2.controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[gazebo-11] [WARN] [1738189685.356751242] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-1] [WARN] [1738189685.805195614] [ur1.controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-6] [WARN] [1738189685.806131753] [ur2.controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[gazebo-11] [WARN] [1738189686.356939237] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[gazebo-11] [WARN] [1738189686.358347284] [gz_ros_control]: Waiting RM to load and initialize hardware...
[ros2_control_node-1] [WARN] [1738189686.805151914] [ur1.controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[ros2_control_node-6] [WARN] [1738189686.806240723] [ur2.controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
[gazebo-11] [WARN] [1738189687.356987524] [controller_manager]: Waiting for data on 'robot_description' topic to finish initialization
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)

```

Thank you in advance!

christophfroehlich, 2025-02-01:
Why do you need two different controller_managers? You can have two namespaced robots in the same CM. Otherwise you have to instantiate two plugins, but there seem to be [issues](https://github.com/ros-controls/gz_ros2_control/issues/196) as well.

augustinoh, 2026-01-21:
> I found it is a gz control2 control problem, that the namespace of the robot should be defined in the plugin.

Hi XHao, could you provide some code to explain this or more detail ? I am trying to do a dual arm simulation with each robots having their own controller, i am struggling so i would really appreciate some help.

thank you

XHao1997, 2026-01-28:
> Hi XHao, could you provide some code to explain this or more detail ? I am trying to do a dual arm simulation with each robots having their own controller, i am struggling so i would really [...]

I'll go back to the project this weekend to see how I did it and if it can help you.