ros-controls/gz_ros2_control #495: Gazebo crashes on giving large cmd_vel velocities in mecanum_drive demo

Amronos, 2025-02-02:
**Describe the bug**
On giving huge cmd_vel velocities on the ``/cmd_vel`` the robot starts moving in very weird ways (to be expected if the wheel joints are being allowed to rotate that fast), and then Gazebo eventually crashes. This issue doesn't occur in diff_drive, the joints are simply blocked from rotating that fast. How are the joints blocked from rotating too fast in diff_drive but not in mecanum_drive?

**To Reproduce**
Steps to reproduce the behavior:
1. Run 
```
ros2 launch gz_ros2_control_demos mecanum_drive_example.launch.py
```
2. Open another terminal and run
```
ros2 topic pub --rate 10 /cmd_vel geometry_msgs/msg/TwistStamped "
twist:
  linear:
    x: 0.0
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 10000.0"
```

**Expected behavior**
The joints should be blocked from rotating that fast and the robot should move at normal speeds in normal ways.

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version Jazzy & Gazebo Harmonic
- I have built ``gz_ros2_control`` from source.

christophfroehlich, 2025-02-03:
the DiffDriveController has an integrated speed limiter, which is configured as follows
https://github.com/ros-controls/gz_ros2_control/blob/789819147a52db8dd7cbf0c18857cfe42911891a/gz_ros2_control_demos/config/diff_drive_controller.yaml#L35-L52

the MecanumDriveController hasn't yet, but we plan to create a TwistLimiterController, which can be chained in front of other controllers with twist input.

closed 2025-02-03 by christophfroehlich

Amronos, 2025-02-03:
Ok, thanks a lot!