ros-controls/gz_ros2_control #503: Argument parser error: Couldn't parse parameter override rule

Jelmerdw, 2025-02-19:
**Describe the bug**
From commit e464e0fd645acfb90b36e45fa44f3702302e9c9e, I experience this error:

`Argument parser error: Couldn't parse parameter override rule: '--param robot_description:=<...> `

When using a xacro with colons in comment lines. This seems to be the exact same issue as described here:
https://github.com/ros-controls/gazebo_ros2_control/issues/295

I do not experience this issue on commits before the commit mentioned above.

mosmz95, 2025-02-19:
I got the same issue after the recent update

[image]

christophfroehlich, 2025-02-19:
I see the problem. Is there a way to wrap the robot_description string? otherwise we'll have to revert #477

sharpclaws, 2025-02-19:
I'm seeing the same issue

christophfroehlich, 2025-02-19:
#505 is a simple fix, do you know if there are other bad characters for the parameter parsing rule?

sharpclaws, 2025-02-20:
Stripping out any colons from comments in the XML did allow me to work around this issue. Note that it's only colons immediately followed by whitespace that are problematic, so `` is fine, but `` is not.
Helpfully, this means that just commenting out something like `<mesh filename="package://my_package/my_mesh.stl" />` will not cause a problem.
Having to search all comments and modify things like `` wasn't a great option. The other workaround is to compile from source before #477 was merged.

ahcorde, 2025-02-20:
There is a new PR fixing this https://github.com/ros-controls/gz_ros2_control/pull/505

Can you try this please ?

mosmz95, 2025-02-20:
I removed all the comments and the problem was solved, also as @sharpclaws said, I tried to keep the comments and only removed the immediate spaces after the colons and it worked. 
thanks

Jelmerdw, 2025-02-20:
> There is a new PR fixing this [#505](https://github.com/ros-controls/gz_ros2_control/pull/505)
> 
> Can you try this please ?

This seems to fix it, nice work!

closed 2025-02-20 by saikishor