ros-controls/gz_ros2_control #504: Odometry TF Frame Missing in ackermann_drive_example.launch.py

ArnoJoosen, 2025-02-19:
**Describe the bug**
While creating a simulation for my robot, I noticed that the odom frame was not present in the TF tree when visualizing the robot in RViz. To investigate further, I tested the ackermann_drive_example.launch.py demo and observed the same behavior: the odom frame was missing from the TF tree, although the odometry data was published as a topic. However, when using the diff_drive_example.launch.py demo, the odom frame was correctly added to the TF tree, allowing the robot's position to be visualized in RViz.

**Steps**
Steps I used:
1. Install ros2_control Packages: `sudo apt install ros-rolling-ros2-control ros-rolling-ros2-controllers`
2. Install gz-ros2-control: `sudo apt isntall ros-jazzy-gz-ros2-control`
3. Clone the gz_ros2_control repository ROS workspace: `git clone -b jazzy https://github.com/ros-controls/gz_ros2_control.git`
4. Move gz_ros2_control_demos to ROS Workspace src.
5. Build Your ROS Workspace: `colcon build --symlink-install`
6. Launch the ackermann_drive_example.launch.py Demo: `ros2 launch gz_ros2_control_demos ackermann_drive_example.launch.py`
    - Gazebo starts up and the robot moves using the command `ros2 run gz_ros2_control_demos example_ackermann_drive`
    - when i run `ros2 topic list` i get:
    ```bash
    /ackermann_steering_controller/controller_state
    /ackermann_steering_controller/odometry
    /ackermann_steering_controller/reference
    /ackermann_steering_controller/tf_odometry
    /ackermann_steering_controller/transition_event
    /clock
    /diagnostics
    /dynamic_joint_states
    /joint_state_broadcaster/transition_event
    /joint_states
    /parameter_events
    /robot_description
    /rosout
    /tf
    /tf_static
    ```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
`ros2 run tf2_tools view_frames`:
[image]

**Environment:**
 - OS: Ubuntu 24.04.1 docker container
 - ROS version: Jazzy

christophfroehlich, 2025-02-20:
please test and review #506

ArnoJoosen, 2025-02-20:
This seems to fix it. Thank you for the quick response!

PR #506 (2025-02-20): Remap to tf to `/tf`
Fixes #504
PR #507 (2025-02-20): Remap to tf to `/tf` (backport #506)
Fixes #504 <hr>This is an automatic backport of pull request #506 done by [Mergify](https://mergify.com).
closed 2025-02-20 by ahcorde