ros-controls/gz_ros2_control #512: Failing to load the controller.yaml

javijurado, 2025-03-12:
**Describe the bug**
I have a package call robot_description with all the urdf, config file etc (all necessary to run gz_ros2_control_demos). When I tried to load the controller on my simulation i got the following error

```bash
[gazebo-2] [INFO] [1741787180.745733094] [controller_manager]: Resource Manager has been successfully initialized. Starting Controller Manager services...
[gazebo-2] [INFO] [1741787180.950502669] [controller_manager]: Loading controller : 'joint_state_broadcaster' of type 'joint_state_broadcaster/JointStateBroadcaster'
[gazebo-2] [INFO] [1741787180.950546700] [controller_manager]: Loading controller 'joint_state_broadcaster'
[gazebo-2] [INFO] [1741787180.954088329] [controller_manager]: Controller 'joint_state_broadcaster' node arguments: --ros-args --params-file --param use_sim_time:=true 
[gazebo-2] [ERROR] [1741787180.954242372] [controller_manager]: 111 Caught exception of type : N6rclcpp10exceptions22RCLInvalidROSArgsErrorE while initializing controller 'joint_state_broadcaster': failed to parse arguments: Couldn't parse params file: '--params-file --param'. Error: Error opening YAML file, at /build/src/ros2/rcl/rcl_yaml_param_parser/src/parser.c:271, at /build/src/ros2/rcl/rcl/src/rcl/arguments.c:415
[spawner-5] [FATAL] [1741787180.955017408] [spawner_joint_state_broadcaster]: Failed loading controller joint_state_broadcaster
[ERROR] [spawner-5]: process has died [pid 49197, exit code 1, cmd '/home/javier/workspace/KION/kion-ros2-rt/install/lib/controller_manager/spawner joint_state_broadcaster --ros-args'].
```

but if instead of that I create a symlink to the same config file but at the level of src folder works.

Path that fail:

` <parameters>/home/javier/workspace/TEST/test-ros2-rt/src/test_nonfix/robot_description/forklift_1252/config/controllers.yaml</parameters>`

Path that works and is a symlink to the previous one:
` <parameters>/home/javier/workspace/TEST/test-ros2-rt/src/test_controllers.yaml</parameters>`

**Environment (please complete the following information):**
 - OS: Debian:12-slim container
- Everything has been build from source
- ROS: jazzy

If something is not clear, i can explain better. Also if some data is needed, I can put it here

christophfroehlich, 2025-03-12:
can you share the URDF section where you load the gz_ros2_control plugin, and the spawner command for the controller from your launchfile?

javijurado, 2025-03-12:
First of all,
Thanks @christophfroehlich for your help on this
- model..urdf.xacro
robot_description lives on: /home/javier/workspace/KION/kion-ros2-rt/src/kion_fix/ros2_prototype_jdm/src/robot_description/forklift_1252/urdf/fl_1252_description.urdf.xacro

This doesn't work
``` 
<gazebo>
    <plugin filename="gz_ros2_control-system" name="gz_ros2_control::GazeboSimROS2ControlPlugin">
      <ros>
        <remapping>ackermann_steering_controller/odometry:=odom</remapping>
        <remapping>ackermann_steering_controller/tf_odometry:=tf</remapping>
        <remapping>ackermann_steering_controller/reference:=cmd_vel_nav</remapping>
      </ros>
      <parameters>$(find robot_description)/forklift_1252/config/controllers.yaml</parameters>
    </plugin>
  </gazebo>
```
This work
``` 
<gazebo>
    <plugin filename="gz_ros2_control-system" name="gz_ros2_control::GazeboSimROS2ControlPlugin">
      <ros>
        <remapping>ackermann_steering_controller/odometry:=odom</remapping>
        <remapping>ackermann_steering_controller/tf_odometry:=tf</remapping>
        <remapping>ackermann_steering_controller/reference:=cmd_vel_nav</remapping>
      </ros>
      <parameters>/home/javier/workspace/KION/kion-ros2-rt/src/controllers.yaml</parameters>
    </plugin>
  </gazebo>
```

- launch
```
joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["joint_state_broadcaster"],
    )

...

          RegisterEventHandler(
                event_handler=OnProcessExit(
                    target_action=gz_spawn_entity,
                    on_exit=[joint_state_broadcaster_spawner],
                )
            ),
            gz_spawn_entity,

```

christophfroehlich, 2025-03-13:
I was not able to reproduce your issue. If the parameter file is not found, then the controller_manager itself won't start.

I don't know where the empty parameter file string comes from, normally this looks like: 
`[gazebo-1] [INFO] [1741897897.779542720] [controller_manager]: Controller 'joint_state_broadcaster' node arguments: --ros-args --params-file /workspaces/ros2_rolling_ws/install/gz_ros2_control_demos/share/gz_ros2_control_demos/config/cart_controller_position.yaml --param use_sim_time:=true `

Can you try to reproduce your issue with the gz_ros2_control_demos?

se1exin, 2025-03-16:
I just had the exact same issue. Coincidentally my package name was also `robot_description`. Renaming my package to something else (`robot_tank_desc`) fixed the issue.

I'm still learning ROS2 so I might be way off, but perhaps `robot_description` being used as the robot's description topic in ROS is causing some kind of conflict? 🤷

thedevmanek, 2025-03-23:
I can confirm that this problem still exists ( using humble latest).
for some reason it is not able to dereference file:// absolute path works as expected but using

javijurado, 2025-03-24:
@christophfroehlich Thanks for your input on this

> Can you try to reproduce your issue with the gz_ros2_control_demos?

I will try to reproduce on the demos and let you know but some people are having the same issue

christophfroehlich, 2025-03-24:
Whoever can help us reproducing the issue will help us fixing it (if it's even a bug in this plugin)

thedevmanek, 2025-03-27:
I was able to run the demos(they worked) and i realised that they use cmake. I am using python does that make a difference. Adding full path works which means that the problem is with $(find pkg) I am not able to use it for meshes as well.Is there a way to debug this?

danilo-schneider, 2025-05-21:
For some reason, only absolute paths works on <parameters> tag. Everything I tried gave me Argument parser error...