ros-controls/gz_ros2_control #513: Difficulty/problem launching ackermann_drive_example.launch.py

tommyleekaelin, 2025-03-12:
**Describe the bug**
I'm getting an error when launching the `ackermann_drive_example.launch.py.` I'm fairly new to robotics and not sure if I've got something completely wrong.

**To Reproduce**
Steps to reproduce the behavior:
1. `colcon build`
2. `source install/setup.bash`
3. Launch `ros2 launch gz_ros2_control_demos ackermann_drive_example.launch.py` 
4. Error: `[spawner-6] spawner: error: unrecognized arguments: --controller-ros-args -r /ackermann_steering_controller/tf_odometry:=/tf``[ERROR] [spawner-6]: process has died [pid 3111, exit code 2, cmd '/opt/ros/jazzy/lib/controller_manager/spawner ackermann_steering_controller --param-file /workspace/install/gz_ros2_control_demos/share/gz_ros2_control_demos/config/ackermann_drive_controller.yaml --controller-ros-args -r /ackermann_steering_controller/tf_odometry:=/tf --ros-args'].`

**Expected behavior**
I think that the ackermann_steering_controller should be configured and activated so that the robot moves when I run `ros2 run gz_ros2_control_demos example_ackermann_drive`

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04.1 LTS noble
 - Version Jazzy
- (working in Docker. Image: unfrobotics/docker-ros2-jazzy-gz-rviz2:latest)

**Additional context**
I tried it in the jazzy and rolling branch.

christophfroehlich, 2025-03-13:
can you check the ros2_control version please? via `ros2 pkg xml -t version ros2_control` for example

tommyleekaelin, 2025-03-13:
yes, I currently have version 4.27.0

christophfroehlich, 2025-03-13:
When I launch the demo, I get

> [gazebo-2] [INFO] [1741905738.362286409] [controller_manager]: Controller 'ackermann_steering_controller' node arguments: --ros-args --params-file /workspaces/ros2_rolling_ws/install/gz_ros2_control_demos/share/gz_ros2_control_demos/config/ackermann_drive_controller.yaml --params-file /workspaces/ros2_rolling_ws/install/gz_ros2_control_demos/share/gz_ros2_control_demos/config/ackermann_drive_controller.yaml --param use_sim_time:=true --ros-args -r /ackermann_steering_controller/tf_odometry:=/tf

which version of gz_ros2_control and of controller_manager have you installed/compiled? 

`--controller-ros-args` was added with

- ros2_control 4.22.0 https://github.com/ros-controls/ros2_control/pull/1713
- and with gz_ros2_control_demos on jazzy branch,  https://github.com/ros-controls/gz_ros2_control/pull/507

tommyleekaelin, 2025-03-18:
so I have for
- gz_ros2_control: 1.2.8
- controller_manager: 4.20.0

christophfroehlich, 2025-03-18:
If you compile it from source, you have to use the correct versions of demos, the plugins, and ros2_control. 
Your versions are from [last November](https://discourse.ros.org/t/new-packages-for-jazzy-jalisco-2024-11-21/40763). Can't you just update the system? (`sudo apt-get update && sudo apt-get upgrade`)
If you use gz_ros2_control from binaries, you can also install gz_ros2_control_demos from binaries. 

If you compile the demos from source with your old versions, you have to use the demos from the same tag, which is https://github.com/ros-controls/gz_ros2_control/releases/tag/1.2.8

closed 2025-03-18 by christophfroehlich