ros-controls/gz_ros2_control #529: Loading parameters from .sdf file

DoppiaEffe94, 2025-03-26:
Hello!

I'm opening this feature request since I've noticed in the [gz_ros2_control_plugin.cpp](https://github.com/ros-controls/gz_ros2_control/blob/8c826da3a2b38d72c5f67c40908f12460d1fb0a4/gz_ros2_control/src/gz_ros2_control_plugin.cpp), and in particular in the lines [L283-L288](https://github.com/ros-controls/gz_ros2_control/blob/8c826da3a2b38d72c5f67c40908f12460d1fb0a4/gz_ros2_control/src/gz_ros2_control_plugin.cpp#L283-L288) that the code is expected to have a parameter file, otherwise it exits.

Since it's possible that sometimes the user wishes to load the parameters from the .sdf file directly, I was wondering if you may take into account the possibility to remove these lines by trading them with a **check of the controller "type"**, for example.

If so, I may try to open a pull request and find a solution with you.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

christophfroehlich, 2025-03-26:
> Since it's possible that sometimes the user wishes to load the parameters from the .sdf file directly,

Which parameters do you think about? The check here is for the [ros2-control yaml files](https://github.com/ros-controls/gz_ros2_control/blob/rolling/gz_ros2_control_demos/config/cart_controller_position.yaml), which are necessary for the controller_manager.

However, we could set the update_rate parameter explicitly from the plugin, then create parameter yamls only for the controllers, passing them to the spawners with the [--param-file option](https://github.com/ros-controls/gz_ros2_control/blob/77663df1977d34ffef85d30e3d8924dda8fc6bc6/gz_ros2_control_demos/launch/cart_example_position.launch.py#L77-L78).

christophfroehlich, 2025-03-28:
But in this case you will need a nested structure to allow for any number of controllers. I don't see the advantage in doing so.

The only thing which might be fine is to only set the update_rate to the same as the gazebo one in case of a missing parameter file and proceed. Then any other controllers have to be loaded by the spawners.

DoppiaEffe94, 2025-04-10:
All right, it sounds reasonable and sure, it won't work in case of multiple controllers.
Then, I'm going to close this issue.

If you still want to set the update_rate as the gazebo one, in case of missing, let me know and I will open a PR.

Thank you for your time and fast response!

closed 2025-04-10 by DoppiaEffe94

christophfroehlich, 2025-04-11:
I'm not sure about the update rate, because you'll need it anyways if one switches to hardware using ros2_control_node 🤔