ros-controls/gz_ros2_control #532: Launching ackermann_drive_example.launch.py fails

louietouie, 2025-03-28:
When launching the example via `ros2 launch gz_ros2_control_demos ackermann_drive_example.launch.py`, the `ackermann_steering_controller` fails to load with the error below.

`
[ERROR] [spawner-6]: process has died [pid 379893, exit code 2, cmd '/opt/ros/jazzy/lib/controller_manager/spawner ackermann_steering_controller --param-file /home/_/_/_/gz_ros2_control/install/gz_ros2_control_demos/share/gz_ros2_control_demos/config/ackermann_drive_controller.yaml --controller-ros-args -r /ackermann_steering_controller/tf_odometry:=/tf -r /ackermann_steering_controller/reference:=/cmd_vel --ros-args'].
`

I fixed this on my own machine by combining line 81 and 82 of the launch file.

`
 '--controller-ros-args',
 '-r /ackermann_steering_controller/tf_odometry:=/tf -r /ackermann_steering_controller/reference:=/cmd_vel'
`

When this is done, the commands are called in the correct order (without `--ros-args` after the `-r` flags as shown in the error)

`--ros-args -r /ackermann_steering_controller/tf_odometry:=/tf -r /ackermann_steering_controller/reference:=/cmd_vel `

I'm new to Gazebo so I'm not too confident, but if I'm correct about this I can definitely submit a PR.

christophfroehlich, 2025-03-28:
On which branch/distro are you experiencing this error?

louietouie, 2025-03-28:
This is on Jazzy, Ubuntu 24.04.

christophfroehlich, 2025-03-28:
You are right, I can reproduce your issue on rolling.

What does not work is the current HEAD

```python
        arguments=['ackermann_steering_controller',
                   '--param-file',
                   robot_controllers,
                   '--controller-ros-args',
                   '-r /ackermann_steering_controller/tf_odometry:=/tf',
                   '-r /ackermann_steering_controller/reference:=/cmd_vel'
                   ],
```
or
```python
        arguments=['ackermann_steering_controller',
                   '--param-file',
                   robot_controllers,
                   '--controller-ros-args',
                   '-r /ackermann_steering_controller/tf_odometry:=/tf',
                   '--controller-ros-args',
                   '-r /ackermann_steering_controller/reference:=/cmd_vel'
                   ],

```
@saikishor should it? Is this a bug in the spawner, or is the fix from @louietouie correct?

saikishor, 2025-03-29:
Hello guys!

Yes, you are right. I thought like parsing everything together. We cannot do the first case as you mentioned @christophfroehlich and the second case can be done. It is simple, just adding an append in the argparse

christophfroehlich, 2025-03-29:
@louietouie please open a PR with your proposal to fix this immediately. For the long run, I added the possibility for the second option above, see the linked PR.