ros-controls/gz_ros2_control #551: Manipulator gravity torques not matching Dart (gz-physics) getGravityForces()

qleonardolp, 2025-04-15:
**Describe the bug**
Hello. I've been developing a ros2 controller using computed torque-control. My goal is to develop an impedance controller. However, I'm facing an issue to implement the gravity compensation (GC). I'm trying to leverage the capability of DART library to compute the manipulator Jacobian, inertia tensor, Coriolis and Gravity effects in joint space. Also, in this way both the simulation (Gazebo) and my controller uses the same robot description (URDF).

One can see how to implement the gravity compensation as present in one of the DART [examples](https://github.com/dartsim/dart/blob/9715990b4bbe54d7ca02ca980b14623704a3a2ff/examples/operational_space_control/main.cpp#L107):

```c++
    Eigen::VectorXd C = mRobot->getGravityForces();

    mRobot->setForces(C);
```

However, the values returned does not match with the torques at the joints for the same manipulator configuration. I could input the torques manually through a ForwardCommandController and the values are diferent.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch my simulation `ur5_simulation.launch.py`
2. Check joint efforts in `/joint_states`
3. Activate my controller `ros2 control set_controller_state impedance_controller active`
4. See error 

**Expected behavior**
When writing on the effort command interfaces the values from the `getGravityForces()` method the robot stand still.

**Screenshots**

Efforts before activating my controller (I guess it is the efforts that the gz_ros2_control sets to hold position):
[image]

These values makes some sense for this spatial configuration:

[image]

Efforts when activate my controller:

[image]

**Environment (please complete the following information):**
 - OS: Ubuntu Noble
 - Version Jazzy
- my repo: https://github.com/qleonardolp/ros2_impedance_controller

**Additional context**
let me konw If this issue/discussion is more suitable to Dart repository.

christophfroehlich, 2025-04-16:
I never checked the effort values as you did with modern gazebo. I'd suggest you use a trivial robot (maybe the RRbot of the ros2_control_demos), and calculate the effort manually to compare it?

saikishor, 2025-04-17:
> I never checked the effort values as you did with modern gazebo. I'd suggest you use a trivial robot (maybe the RRbot of the ros2_control_demos), and calculate the effort manually to compare it?

That's a good idea

qleonardolp, 2025-04-22:
> I never checked the effort values as you did with modern gazebo. I'd suggest you use a trivial robot (maybe the RRbot of the ros2_control_demos), and calculate the effort manually to compare it?

Thanks for this ideia. I checked it and could confirm it is a Dart related issue.

closed 2025-04-22 by qleonardolp