ros-controls/gz_ros2_control #567: ros2_control_node incompatible with gz_ros2_control/GazeboSimSystem

SphericalCowww, 2025-04-26:
**Describe the bug**
This is for Jazzy with Gazebo Harmonics on Ubuntu 24.04.2 LTS on Raspberry Pi 5. 
When trying to use `ros2_control_node` with `gz_ros2_control/GazeboSimSystem`, I got the following error:
```
[ros2_control_node-2] [ERROR] [1745662912.685450215] [controller_manager.resource_manager]: Caught exception of type: N9pluginlib20LibraryLoadExceptionE while loading hardware: According to the loaded plugin descriptions, the class gz_ros2_control/GazeboSimSystem with base class type hardware_interface::SystemInterface does not exist. Declared types are  mock_components/GenericSystem
```
Note that the Gazebo can be launched with `gz_ros2_control::GazeboSimROS2ControlPlugin` that can use `gz_ros2_control::GazeboSimSystemInterface` instead of `hardware_interface::SystemInterface`, so I feel like this is a quick exception error.

**To Reproduce**
Minimal urdf file (called `arduinobot_debug.urdf.xacro` under `robot_description/urdf`):

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="arduinobot">
    <ros2_control type="system" name="RobotSystem">
        <hardware>
            <plugin>gz_ros2_control/GazeboSimSystem</plugin>
        </hardware>
    </ros2_control>

    <link name="world"/>
</robot>
```

Minimal launch file:

```
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node
from launch_ros.parameter_descriptions import ParameterValue
from launch.substitutions import Command, LaunchConfiguration
##############################################################################################################################
def generate_launch_description():
    ld = LaunchDescription() 
    robot_description = ParameterValue(
        Command([
            "xacro",
            " ",
            os.path.join(
                get_package_share_directory("robot_description"),
                "urdf",
                "arduinobot_debug.urdf.xacro",
            ),
        ]),
        value_type=str,
    )

    robot_state_publisher = Node(
        package   ="robot_state_publisher",
        executable="robot_state_publisher",
        parameters=[{
            "robot_description": robot_description,
            "use_sim_time": True,    
        }],
    )
    controller_manager = Node(
        package   ="controller_manager",
        executable="ros2_control_node",
        parameters=[
            {"robot_description": robot_description},
            os.path.join(
                get_package_share_directory("robot_controller"),
                "config",
                "arduinobot_controller.yaml",
            ),
        ],
    )
    ld.add_action(robot_state_publisher)
    ld.add_action(controller_manager)
    return ld
```

Relevant `package.xml`:

```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>robot_description</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="tinglin194@gmail.com">kali</maintainer>
  <license>TODO: License declaration</license>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <exec_depend>urdf</exec_depend>
  <exec_depend>xacro</exec_depend>
  <exec_depend>robot_state_publisher</exec_depend>
  <exec_depend>joint_state_publisher_gui</exec_depend>
  <exec_depend>rviz2</exec_depend>
  <exec_depend>ros2launch</exec_depend>
  <exec_depend>gazebo_ros</exec_depend>
  <exec_depend>ros_gz_sim</exec_depend>
  <exec_depend>ros_gz_bridge</exec_depend>
  <exec_depend>gz_ros2_control</exec_depend>

  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
```

**Expected behavior**
I expect to use `ros2_control_node` to control the robot in Gazebo.
Thanks in advance.

**Environment (please complete the following information):**
- OS: Ubuntu 24.04.2 LTS
- Version: Jazzy with Gazebo Harmonics
- Arm chip on Raspberry Pi 5
- Packages installed via:

```
sudo apt install  ros-jazzy-ros2-control ros-jazzy-controller-manager ros-jazzy-ros2-controllers ros-jazzy-gz-ros2-control
```

christophfroehlich, 2025-04-26:
Have a look at the demos, we never use your pattern there to launch a ros2_control_node.

ros2_control_node is not compatible with gz_ros2_control, because the plugin creates its own instance of the controller_manager.

Why do you want to launch that?

SphericalCowww, 2025-04-26:
Hello, thank you for your reply. This project is directly from an Udemy tutorial. And after several checks, together with a demo, I believe I understand the lecturer's assessment. The `gz_ros2_control` initializes with the following:

```
Hardware Component 1
	name: RobotSystem
	type: 
	plugin name: 
	state: id=3 label=active
	command interfaces
		joint_1/position [available] [unclaimed]
		joint_2/position [available] [unclaimed]
		joint_3/position [available] [unclaimed]
		joint_4/position [available] [unclaimed]

```
And here is a slightly longer version of the launch code just to claim the joints:

```
  controller_manager = Node(
        package   ="controller_manager",
        executable="ros2_control_node",
        parameters=[
            {"robot_description": robot_description},
            os.path.join(
                get_package_share_directory("robot_controller"),
                "config",
                "arduinobot_controller.yaml",
            ),
        ],
    )
    arm_controller_spawner = Node(
        package   ="controller_manager",
        executable="spawner",
        arguments = [
            "arm_controller",
            "--controller-manager",
            "/controller_manager",
        ],
    )
```

The question is, why `ros2_control_node`, why not just do `spawner` directly? I think there are two reasons:

1. I think `<plugin>gz_ros2_control/GazeboSimSystem</plugin>` is not fundamentally different from say `<plugin>ros2_control_demo_example_2/DiffBotSystemHardware</plugin>` or `<plugin>mock_components/GenericSystem</plugin>`, so why different treatment, especially when the same code runs just fine when the Udemy lecturer runs it.
2. Without `ros2_control_node`, we will encounter the following bug: [stackexchange](https://robotics.stackexchange.com/questions/115202/failed-to-load-controller-joint-state-broadcaster-params-file-param-error), or:

```
[gazebo-2] [INFO] [1745687884.127667831] [controller_manager]: Controller 'arm_controller' node arguments: --ros-args --params-file --param use_sim_time:=true 
[gazebo-2] [ERROR] [1745687884.130832430] [controller_manager]: Caught exception of type : N6rclcpp10exceptions22RCLInvalidROSArgsErrorE while initializing controller 'arm_controller': failed to parse arguments: Couldn't parse params file: '--params-file --param'. Error: Error opening YAML file, at ./src/parser.c:271, at ./src/rcl/arguments.c:415
```

I have tried several workarounds to avoid this default bugged argument `--params-file --param`. For instance, if I specified the path to `--params-file`, 

```
    arm_controller_spawner = Node(
        package   ="controller_manager",
        executable="spawner",
        arguments = [
            "arm_controller",
            "--controller-manager",
            "/controller_manager",
            "-p",
            os.path.join(
                get_package_share_directory("robot_controller"),
                "config",
                "arduinobot_controller.yaml",
            ),
        ],
    )
```
But it only gets me so far (same error if using spawner directly from the command line terminal):

```
[gazebo-2] [INFO] [1745688936.897100792] [controller_manager]: Controller 'arm_controller' node arguments: --ros-args --params-file --params-file /home/.../install/robot_controller/share/robot_controller/config/arduinobot_controller.yaml --param use_sim_time:=true 
[gazebo-2] [ERROR] [1745688936.898208704] [controller_manager]: Caught exception of type : N6rclcpp10exceptions22RCLInvalidROSArgsErrorE while initializing controller 'arm_controller': failed to parse arguments: Couldn't parse params file: '--params-file --params-file'. Error: Error opening YAML file, at ./src/parser.c:271, at ./src/rcl/arguments.c:415
```
It instead gives this odd `--params-file --params-file` combo. And to my understanding, `ros2_control_node` serves to mitigate this issue. So I believe fixing either point 1 or point 2 would be very helpful for a flexible implementation of `gz_ros2_control` in general. Thank you very much in advance.

christophfroehlich, 2025-04-26:
> 1. I think `<plugin>gz_ros2_control/GazeboSimSystem</plugin>` is not fundamentally different from say `<plugin>ros2_control_demo_example_2/DiffBotSystemHardware</plugin>` or [...]

but it IS fundamentally different, as I said, see  `GazeboSimROS2ControlPlugin::PreUpdate` and `PostUpdate` methods.  That's why it does not work from the ros2_control_node.

regarding 2): have you tried the suggestion from the answers on stack exchange? Please open a different issue for that.

by the way, giving robot_description as parameter to the ros2_control_node is deprecated and not supported [from jazzy](https://control.ros.org/rolling/doc/ros2_control/doc/migration.html#controller-manager) on.

> controller_manager = Node(
>         package   ="controller_manager",
>         executable="ros2_control_node",
>         parameters=[
>             {"robot_description": robot_description}, [...]

SphericalCowww, 2025-04-27:
Do you mean `GazeboSimROS2ControlPlugin::PreUpdate` and `PostUpdate` as in the source code?! Okay, I am unsure which update it refers to and why `gz_ros2_control` is inherently developed to be inconsistent with `ros2_control`, but I will read about it if I have spare time.

And regarding 2), the github link you provided on the [stackexchange](https://robotics.stackexchange.com/questions/115202/failed-to-load-controller-joint-state-broadcaster-params-file-param-error) is down, however, I think the workaround I used was akin to your answer, I believe. The other one is to not name my source folder "robot_description"... I will give it a shot. But yes, this should be another issue.

Thanks for your info.

closed 2025-04-27 by SphericalCowww