ros-controls/gz_ros2_control #568: [Effort controller] Set initial config

RainJCloude, 2025-04-27:
Hello, I would like to use effort controller to command robots my I don't manage to give it an initial config.
I can do that just if I set a position controller setting the joints value in the state interface, but these values are ignored if I used an effort controller, or if I still use a position controller with both position and effort command interfaces.

Do you have any suggestions on how to cope with this problem?
Thanks a lot

christophfroehlich, 2025-04-27:
Do you want to set the [initial effort](https://github.com/ros-controls/gz_ros2_control/blob/5398c37f657544ceecb3fd3be3e548f28d090928/gz_ros2_control/src/gz_system.cpp#L335C12-L335C26), or initial position? 

The latter is not possible if you use effort_command interface. In general gazebo does not support setting initial values from URDF. 
You'll need to use a position controller (claiming position command interface) first to set the position, and then switch to your effort controller. Or you configure a feedback loop (pid_controller for example), which moves your robot to the desired position via effort command interface.

RainJCloude, 2025-04-28:
Thanks, a feedback loop should be the most elegant solution

christophfroehlich, 2025-04-28:
"elegant" depends on the use case here 😁 can we close this issue?

RainJCloude, 2025-04-29:
Sure, thank you so much