ros-controls/gz_ros2_control #577: Gazebo crashes on startup, when the controller manager tries to config the diff_drive controller (ROS Jazzy and Gazebo Harmonic)

michailtam, 2025-05-12:
I am using the ros2_control differential drive controller for Jazzy and want to spawn the robot in Gazebo Harmonic. When Gazebo launch the controller was loaded by the controller manager properly, but when it tries to config the diff-drive-controller the execution outputs the following error message:

`[gazebo-3] [INFO] [1747069202.382875778] [controller_manager]: Loading controller 'diff_controller'
[gazebo-3] [INFO] [1747069202.385938432] [controller_manager]: Controller 'diff_controller' node arguments: --ros-args --params-file /media/mikelap/DATA2/Development/GithubRepos/hospital_robot_ws/hospital_robot_ws/install/hrobot_bringup/share/hrobot_bringup/config/ros2_controllers.yaml -p use_sim_time:=true --params-file /media/mikelap/DATA2/Development/GithubRepos/hospital_robot_ws/hospital_robot_ws/install/hrobot_bringup/share/hrobot_bringup/config/ros2_controllers.yaml --param use_sim_time:=true 
[spawner-10] [INFO] [1747069202.393105863] [spawner_diff_controller]: Loaded diff_controller
[gazebo-3] [INFO] [1747069202.393707351] [controller_manager]: Configuring controller: 'diff_controller'
[gazebo-3] Stack trace (most recent call last) in thread 18821:
[gazebo-3] #13   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[gazebo-3] #12   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x72bb0cf29c3b, in 
[gazebo-3] #11   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x72bb0ce9caa3, in 
[gazebo-3] #10   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x72bb076ecdb3, in 
[gazebo-3] #9    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x72babed1b116, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[gazebo-3] #8    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x72babed08d69, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[gazebo-3] #7    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x72babed05043, in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>)
[gazebo-3] #6    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x72babeddd049, in 
[gazebo-3] #5    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x72bac430452d, in 
[gazebo-3] #4    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x72bac42e24e9, in std::_Function_handler<void (std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >), std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >&&)
[gazebo-3] #3    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x72bac429bb84, in controller_manager::ControllerManager::configure_controller_service_cb(std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >)
[gazebo-3] #2    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x72bac4291da2, in controller_manager::ControllerManager::configure_controller(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[gazebo-3] #1    Object "/opt/ros/jazzy/lib/libcontroller_interface.so", at 0x72bac40bc47c, in 
[gazebo-3] #0    Object "/opt/ros/jazzy/lib/libcontroller_interface.so", at 0x72bac40bd173, in 
[gazebo-3] Segmentation fault (Address not mapped to object [(nil)])`

The output shows, that the diff_controller gets loaded properly 

> [spawner-10] [INFO] [1747069202.393105863] [spawner_diff_controller]: Loaded diff_controller

but when the controller manager tries to configure it, Gazebo crashes with the following output

>[gazebo-3] [INFO] [1747069202.393707351] [controller_manager]: Configuring controller: 'diff_controller'
[gazebo-3] Stack trace (most recent call last) in thread 18821:
[gazebo-3] #13   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[gazebo-3] #12   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x72bb0cf29c3b, in 
[gazebo-3] #11   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x72bb0ce9caa3, in 
[gazebo-3] #10   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x72bb076ecdb3, in 
[gazebo-3] #9    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x72babed1b116, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[gazebo-3] #8    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x72babed08d69, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[gazebo-3] #7    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x72babed05043, in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>)
[gazebo-3] #6    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x72babeddd049, in 
[gazebo-3] #5    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x72bac430452d, in 
[gazebo-3] #4    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x72bac42e24e9, in std::_Function_handler<void (std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >), std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >&&)
[gazebo-3] #3    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x72bac429bb84, in controller_manager::ControllerManager::configure_controller_service_cb(std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >)
[gazebo-3] #2    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x72bac4291da2, in controller_manager::ControllerManager::configure_controller(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[gazebo-3] #1    Object "/opt/ros/jazzy/lib/libcontroller_interface.so", at 0x72bac40bc47c, in 
[gazebo-3] #0    Object "/opt/ros/jazzy/lib/libcontroller_interface.so", at 0x72bac40bd173, in 
[gazebo-3] Segmentation fault (Address not mapped to object [(nil)])

My code snippets related to this issue are:

### Code

**launch file**
`diff_drive_controller_cmd = Node(
       package='controller_manager',
       executable='spawner',
       arguments=[
          'diff_controller', '--param-file', controller_file
     ])`

**controller_file (YAML)**
`
controller_manager:
  ros__parameters:
    update_rate: 100 # 100 Hz
    use_sim_time: true

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    diff_controller:
[... 55 lines cut ...]
        <mu2>1000000000000000.0</mu2>
        <kp>1000000000000.0</kp>
        <kd>10.0</kd>
        <minDepth>0.001</minDepth>
        <maxVel>0.1</maxVel>
        <fdir1>1 0 0</fdir1>
    </gazebo>

    <gazebo reference="WheelB_link">
        <mu1>0.1</mu1>
        <mu2>0.1</mu2>
        <kp>1000000.0</kp>
        <kd>100.0</kd>
        <minDepth>0.001</minDepth>
        <maxVel>1.0</maxVel>
    </gazebo>

    
    <gazebo>
        <plugin filename="gz_ros2_control-system" name="gz_ros2_control::GazeboSimROS2ControlPlugin">
            <parameters>$(find hrobot_bringup)/config/ros2_controllers.yaml</parameters>
            
        </plugin>
    </gazebo>
`

michailtam, 2025-05-12:
[Failed to load system plugin gz::sim::Systems::DiffDrive #2715
](#2715)

closed 2025-05-14 by michailtam

christophfroehlich, 2025-05-15:
do you mind  sharing your solution for other users, which might run into the same issue in the future?