ros-controls/gz_ros2_control #578: mecanum_drive_example not moving lateral a linear velocity in y

javijurado, 2025-05-14:
**Describe the bug**
mecanum_drive_example not moving lateral a linear velocity in y

**To Reproduce**
See the video
https://github.com/user-attachments/assets/c1504160-fa40-4788-bcc7-4d28b0636907

**Expected behavior**
I expected to move lateral
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Docker container running ros2 humble

Am I doing something wrong?

NickZhou11000, 2025-08-26:
The demo code for humble is not going to work for Ignition Gazebo Fortress, which is recommended for ROS2 Humble.
To make the simulation successful,  you need to modify the urdf file `test_mecanum_drive.xacro.urdf`.

Apparently that Ignition Gazebo had changed the api for `<fdir1>`. \
For Fortress and before, it use `<fdir1 ignition:expressed_in='base_link'>1 -1 0</fdir1>` \
For Garden and after, it use `<fdir1 gz:expressed_in='base_link'>1 -1 0</fdir1>`

Following are demo code for Fortress and older:
```xml
<robot name="mecanum_vehicle" xmlns:xacro="http://www.ros.org/wiki/xacro" xmlns:ignition="http://gazebosim.org/schema">

  <gazebo reference='**_**_wheel'>
    <collision>
      <surface>
        <friction>
          <ode>
            <mu>1.0</mu>
            <mu2>0.0</mu2>
            <fdir1 ignition:expressed_in="base_link">1.0 -1.0 0.0</fdir1>
          </ode>
        </friction>
      </surface>
    </collision>
  </gazebo>

</robot>

```

Following are demo code for Garden and newer:
```xml
<robot name="mecanum_vehicle" xmlns:xacro="http://www.ros.org/wiki/xacro" xmlns:gz="http://gazebosim.org/schema">

  <gazebo reference='**_**_wheel'>
    <collision>
      <surface>
        <friction>
          <ode>
            <mu>1.0</mu>
            <mu2>0.0</mu2>
            <fdir1 gz:expressed_in="base_link">1.0 -1.0 0.0</fdir1>
          </ode>
        </friction>
      </surface>
    </collision>
  </gazebo>

</robot>

```
Don't forget the `xmlns:ignition="http://gazebosim.org/schema"` or `xmlns:gz="http://gazebosim.org/schema"` at the beginning

christophfroehlich, 2025-08-26:
Thanks for pointing this out. 

> The demo code for humble is not going to work for Ignition Gazebo Fortress, which is recommended for ROS2 Humble.
> To make the simulation successful, you need to modify the urdf file `test_mecanum_drive.xacro.urdf`.

I just don't understand the versions here, humble+fortress is the released combination, where the current URDF should be fine according to your snippets?

Do you mind opening a PR against the rolling branch, which will be backported to kilted and jazzy? on humble we would have to create a xacro if/unless branch or two separate URDF files.