ros-controls/gz_ros2_control #605: sdf example for humble

Ali-J-Fesharaki, 2025-06-08:
when I copy paste launch files example from  jazzy to humble we have some bugs and It's related to ros2-control node 
please refine that and add some sdf example for humble

main error:
[ruby $(which ign) gazebo-1] [ERROR] [1749384721.639747104] [gz_ros2_control]: Error parsing URDF in gz_ros2_control plugin, plugin not active : the robot tag is not root element in URDF

the whole log:
<pre>ros2 launch gz_ros2_control_demos diff_drive_example.launch.py description_format:=sdf
[INFO] [launch]: All log files can be found below /home/ajf/.ros/log/2025-06-08-15-41-59-215954-tayeb-Gaming-240636
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [240638]
[INFO] [parameter_bridge-2]: process started with pid [240641]
[INFO] [create-3]: process started with pid [240643]
[INFO] [robot_state_publisher-4]: process started with pid [240646]
[create-3] [INFO] [1749384719.508922562] [ros_gz_sim]: Requesting list of world names.
[parameter_bridge-2] [INFO] [1749384719.520054095] [ros_gz_bridge]: Creating GZ-&gt;ROS Bridge: [/clock (gz.msgs.Clock) -&gt; /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[robot_state_publisher-4] [WARN] [1749384719.554075915] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-4] [INFO] [1749384719.555292490] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1749384719.555332710] [robot_state_publisher]: got segment caster
[robot_state_publisher-4] [INFO] [1749384719.555336205] [robot_state_publisher]: got segment chassis
[robot_state_publisher-4] [INFO] [1749384719.555338253] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-4] [INFO] [1749384719.555339980] [robot_state_publisher]: got segment right_wheel
[ruby $(which ign) gazebo-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[create-3] [INFO] [1749384719.911055445] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-3] [INFO] [1749384720.010387884] [ros_gz_sim]: Requested creation of entity.
[create-3] [INFO] [1749384720.010457129] [ros_gz_sim]: OK creation of entity.
[INFO] [create-3]: process has finished cleanly [pid 240643]
[INFO] [spawner-5]: process started with pid [240807]
[spawner-5] [INFO] [1749384720.263543703] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ruby $(which ign) gazebo-1] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-1] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-1] [INFO] [1749384721.629087757] [GazeboSimROS2ControlPlugin]: robot_param_node is robot_state_publisher
[ruby $(which ign) gazebo-1] [INFO] [1749384721.629106697] [GazeboSimROS2ControlPlugin]: robot_param_node is robot_description
[ruby $(which ign) gazebo-1] [INFO] [1749384721.639025444] [gz_ros2_control]: connected to service!! robot_state_publisher asking for robot_description
[ruby $(which ign) gazebo-1] [INFO] [1749384721.639278474] [gz_ros2_control]: Received URDF from param server
[ruby $(which ign) gazebo-1] [INFO] [1749384721.639644894] [gz_ros2_control]: [gz_ros2_control] Fixed joint [chassis_joint] (Entity=24)] is skipped
[ruby $(which ign) gazebo-1] [INFO] [1749384721.639651994] [gz_ros2_control]: [gz_ros2_control] Fixed joint [caster_joint] (Entity=27)] is skipped
[ruby $(which ign) gazebo-1] [ERROR] [1749384721.639747104] [gz_ros2_control]: Error parsing URDF in gz_ros2_control plugin, plugin not active : the robot tag is not root element in URDF
[spawner-5] [WARN] [1749384730.278887691] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-5] [INFO] [1749384730.280381548] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
</pre>

Ali-J-Fesharaki, 2025-06-08:
> when I copy paste launch files example from jazzy to humble we have some bugs and It's related to ros2-control node please refine that and add some sdf example for humble
> 
same issue:
https://github.com/ros-controls/gz_ros2_control/issues/419

> main error: [ruby $(which ign) gazebo-1] [ERROR] [1749384721.639747104] [gz_ros2_control]: Error parsing URDF in gz_ros2_control plugin, plugin not active : the robot tag is not root element in URDF
> 
> the whole log:
> 
> ros2 launch gz_ros2_control_demos diff_drive_example.launch.py description_format:=sdf
> [INFO] [launch]: All log files can be found below /home/ajf/.ros/log/2025-06-08-15-41-59-215954-tayeb-Gaming-240636
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [ruby $(which ign) gazebo-1]: process started with pid [240638]
> [INFO] [parameter_bridge-2]: process started with pid [240641]
> [INFO] [create-3]: process started with pid [240643]
> [INFO] [robot_state_publisher-4]: process started with pid [240646]
> [create-3] [INFO] [1749384719.508922562] [ros_gz_sim]: Requesting list of world names.
> [parameter_bridge-2] [INFO] [1749384719.520054095] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (gz.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
> [robot_state_publisher-4] [WARN] [1749384719.554075915] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
> [robot_state_publisher-4] [INFO] [1749384719.555292490] [robot_state_publisher]: got segment base_link
> [robot_state_publisher-4] [INFO] [1749384719.555332710] [robot_state_publisher]: got segment caster
> [robot_state_publisher-4] [INFO] [1749384719.555336205] [robot_state_publisher]: got segment chassis
> [robot_state_publisher-4] [INFO] [1749384719.555338253] [robot_state_publisher]: got segment left_wheel
> [robot_state_publisher-4] [INFO] [1749384719.555339980] [robot_state_publisher]: got segment right_wheel
> [ruby $(which ign) gazebo-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
> [create-3] [INFO] [1749384719.911055445] [ros_gz_sim]: Waiting messages on topic [robot_description].
> [create-3] [INFO] [1749384720.010387884] [ros_gz_sim]: Requested creation of entity.
> [create-3] [INFO] [1749384720.010457129] [ros_gz_sim]: OK creation of entity.
> [INFO] [create-3]: process has finished cleanly [pid 240643]
> [INFO] [spawner-5]: process started with pid [240807]
> [spawner-5] [INFO] [1749384720.263543703] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
> [ruby $(which ign) gazebo-1] libEGL warning: egl: failed to create dri2 screen
> [ruby $(which ign) gazebo-1] libEGL warning: egl: failed to create dri2 screen
> [ruby $(which ign) gazebo-1] [INFO] [1749384721.629087757] [GazeboSimROS2ControlPlugin]: robot_param_node is robot_state_publisher
> [ruby $(which ign) gazebo-1] [INFO] [1749384721.629106697] [GazeboSimROS2ControlPlugin]: robot_param_node is robot_description
> [ruby $(which ign) gazebo-1] [INFO] [1749384721.639025444] [gz_ros2_control]: connected to service!! robot_state_publisher asking for robot_description
> [ruby $(which ign) gazebo-1] [INFO] [1749384721.639278474] [gz_ros2_control]: Received URDF from param server
> [ruby $(which ign) gazebo-1] [INFO] [1749384721.639644894] [gz_ros2_control]: [gz_ros2_control] Fixed joint [chassis_joint] (Entity=24)] is skipped
> [ruby $(which ign) gazebo-1] [INFO] [1749384721.639651994] [gz_ros2_control]: [gz_ros2_control] Fixed joint [caster_joint] (Entity=27)] is skipped
> [ruby $(which ign) gazebo-1] [ERROR] [1749384721.639747104] [gz_ros2_control]: Error parsing URDF in gz_ros2_control plugin, plugin not active : the robot tag is not root element in URDF
> [spawner-5] [WARN] [1749384730.278887691] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
> [spawner-5] [INFO] [1749384730.280381548] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...

Amronos, 2025-06-08:
This is due to ros2_control#1763 not being backported to humble.

closed 2025-09-08 by github-actions[bot]